RobotLocomotion/drake #12128: Improve tetrahedral mesh connectivity for cylinders and boxes.

DamrongGuoy, 2019-10-01:
We should improve mesh connectivity to support Hydroelastic contact model. There are two problems that I found after visualizing the tetrahedral meshes.

1. MakeBoxVolumeMesh() creates a tetrahedral mesh of a box that has some tetrahedra with all vertices on the box boundary. These tetrahedra will define zero extent field in their volume.  This picture shows some of these tetrahedra.
[image]

2. MakeCylinderMesh() creates a tetrahedral mesh of a cylinder that has some tetrahedra spanning two layers. The picture below shows the problematic tetrahedra when we set the clipping plane through the second layer. All other tetrahedra in the first layer were hidden, but the problematic tetrahedra are shown because they span both the first and the second layer. The problematic tetrahedra come in groups of four tetrahedra forming octahedra. I believe when MakeCylinderMesh() share the same SplitOctahedron() code as  MakeSphereVolumeMesh(), we won't have this problem (this task is scheduled in Q4 CY2019). It is not clear how these problematic tetrahedra would affect the discretized extent field.
[image]

CC: @sherm1 @SeanCurtis-TRI @amcastro-tri @joemasterjohn

SeanCurtis-TRI, 2020-04-06:
Why is the cylinder mesh a problem?

amcastro-tri, 2020-04-06:
Good question, I don't see a problem with the tetrahedra being highlighted above.
I wonder though if there are tetrahedra on the top/bottom rims that, as with the box, have all their nodes on the rim and thus a penetration extend field that is zero in the entirety of each tetrahedra. That'd definitely be a big problem.

SeanCurtis-TRI, 2020-04-06:
@amcastro-tri I agree with you -- I likewise assumed that the cylinder corners would/could suffer from the same problem as box corners. Corners being the nuisance.

DamrongGuoy, 2020-10-22:
PR #14216 is merged now. I'll enable the box mesh in hydroelastics. After that, I'll check the cylinder.

PR #14506 (2021-01-12): [geometry] Use Cylinder's meshes with medial axes in hydroelastics.
Fix #12128 "Improve tetrahedral mesh connectivity for cylinders and boxes."

This is how pressure fields would look on the new meshes (drake/geometry/proximity/images):

[image]
[image]

I also verify that it is used in hydroelastics by `contact_surface_rigid_bowl_soft_ball --rigid=cylinder --soft=cylinder` like this:

https://user-images.githubusercontent.com/42557859/104261788-2970e400-543b-11eb-9bd1-37f2257c6e26.mp4

and `--rigid=box --soft=cylinder` like this:

https://user-images.githubusercontent.com/42557859/104261843-4 [... cut]
closed 2021-01-12 by rpoyner-tri