RobotLocomotion/drake #12134: The RgbdSensor in drake only outputs rgba image, should also output rgb image

huihuaTRI, 2019-10-02:
The RgbdSensor in drake does have a RGB image data member, but not output that as a port. The FMK simulation requires a rgb image channel.

SeanCurtis-TRI, 2019-10-02:
I suspect that if a channel needs to be stripped out (i.e., alpha), it shouldn't be done by the sensor, per se, but by a downstream member. 

That's not to say we couldn't add an output port that strips it out, but it would include deprecating current names (e.g., what does `color_image_output_port` *mean* if there are two color ports: RGB and RGBA?)

And if we were to add this additional port, it would *literaly* render *with* alpha and then strip it out. So, it wouldn't be a particular savings in computation to do it locally.

huihuaTRI, 2019-10-03:
Thank you for confirming that the RGB channel is just the first three channels of the RGBA, i.e., alpha channel is independent.  In this case, I agree with you and I can just close this issue.

SeanCurtis-TRI, 2019-10-22:
@huihuaTRI I'm closing this now, as per conversation. You can re-open this if it proves to be premature.

closed 2019-10-22 by SeanCurtis-TRI