RobotLocomotion/drake #12137: Should fix for bazel 1.0.0

EricCousineau-TRI, 2019-10-03:
See PR #12124

Things break for shared libraries (at least our formulation)?

\cc @jwnimmer-tri

jwnimmer-tri, 2019-10-03:
Yes, we need `--incompatible_remove_legacy_whole_archive=false` as a stopgap, and then some `alwayslink = 1` repairs to pydrake later in order to undo that flag.

jwnimmer-tri, 2019-10-17:
Where is the tracking issue to remove `--incompatible_remove_legacy_whole_archive=false` as a requirement in our rcfiles?  Users who consume us like `bazel_drake_external` currently have a pain point where they need to change _their own project_ in order to consume Drake.

jwnimmer-tri, 2019-10-28:
@EricCousineau-TRI The other, more important build system transitions seem to be making good progress.  I'd like to see this issue fixed prior to the v0.12 release (so, merged to master in the next week or so).  If that's not possible, please re-assign this issue to me to work on it.

EricCousineau-TRI, 2019-10-29:
Spoke with @jamiesnape and it'd be feasible, but not in time for 0.12 release. I'll reassign to you for the time being. Thanks!

jwnimmer-tri, 2019-11-07:
For the record, trying to fix this has uncovered several challenges.  I don't think it will make the v0.12.0 release cycle.

PR #12262 (2019-11-21): libdrake: Use alwayslink instead of legacy_whole_archive
As a side-effect, duplicate definitions (e.g., of `main`) are now an error.

Closes #12137.

Relates https://github.com/bazelbuild/bazel/issues/7362.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12262)
closed 2019-11-21 by jwnimmer-tri