RobotLocomotion/drake #12140: In MultibodyPlant, the acceleration bias term APIs are misnamed

sherm1, 2019-10-05:
CalcBiasForJacobianTranslationalVelocity() and CalcBiasForJacobianSpatialVelocity()
should be CalcBiasForTranslationalAcceleration() and CalcBiasForSpatialAcceleration() or something like that. They actually return accelerations, not velocities.

@mitiguy noticed these first and put TODOs in #12135 since deprecation will be needed.

mitiguy, 2020-03-31:
This issue was raised again in PR# 12839.  For consistency and parallel to the currently-implemented Jacobian names and with what is proposed in PR# 12839,  let's consider:

* CalcBiasAngularAcceleration() 
* CalcBiasTranslationalAcceleration() 
* CalcBiasSpatialAcceleration() 
* CalcBiasCenterOfMassTranslationalAcceleration() -- done in PR #12839 

This parallels the currently-implemented Jacobian names:
* CalcJacobianAngularlVelocity()
* CalcJacobianTranslationalVelocity()
* CalcJacobianSpatialVelocity()
* CalcJacobianCenterOfMassTranslationalVelocity()  -- done in PR #12839

sherm1, 2020-03-31:
I like these names. We should be sure to cross-reference in Doxygen though since it won't be obvious which methods are related.

mitiguy, 2020-04-08:
It would be helpful to have acceleration bias notation that parallels the currently-implemented Jacobian notation.  Herein 𝑠 = q̇ or 𝑠 = v.

J𝑠_w_AB_E  = CalcJacobianAngularVelocity( ... )    
Frame B's angular velocity Jacobian in frame A with respect to speeds 𝑠, expressed in frame E.

Example: Jq̇_v_AB_E  or  Jv_v_AB_E
J𝑠_v_ABi_E = CalcJacobianTranslationalVelocity( ... )
J𝑠_v_ACcm_E  = CalcJacobianCenterOfMassTranslationalVelocity()
Point Bi's translational velocity Jacobian in frame A with respect to speeds 𝑠, expressed in frame E.
Point Ccm's translational velocity Jacobian in frame A with respect to speeds 𝑠, expressed in frame E.

J𝑠_V_AB_E = CalcJacobianSpatialVelocity()
Frame B's spatial velocity Jacobian in frame A with respect to speeds 𝑠, expressed in frame E.
---------------------------------------
B𝑠_alpha_AB_E  = CalcBiasAngularAcceleration( ... )  
Frame B's angular acceleration bias in frame A with respect to speeds 𝑠, expressed in frame E.

Example: Bq̇_a_AB_E  or  Bv_a_AB_E
B𝑠_a_ABi_E = CalcBiasTranslationalAcceleration( ... )
B𝑠_a_ACcm_E  = CalcBiasCenterOfMassTranslationalAcceleration()
Point Bi's translational acceleration bias in frame A with respect to speeds 𝑠, expressed in frame E.
Point Ccm's translational acceleration bias in frame A with respect to speeds 𝑠, expressed in frame E.

B𝑠_A_AB_E = CalcBiasSpatialAcceleration()
Frame B's spatial acceleration bias in frame A with respect to speeds 𝑠, expressed in frame E.

mitiguy, 2020-04-27:
Created PR #13063 to rename CalcBiasForJacobianSpatialVelocity() to CalcBiasSpatialAcceleration() and rename CalcBiasForJacobianTranslationalVelocity()  to CalcBiasTranslationalAcceleration().

mitiguy, 2020-05-21:
PR# 13063 was merged and seems to address all the remaining items in this issue except for the method CalcBiasAngularAcceleration().  Thus far, that method has not been needed.  If it is needed, this issue can be re-opened or a new issue can be opened.

closed 2020-05-21 by mitiguy