RobotLocomotion/drake #12151: test_linear_algebra failing with numpy 1.17.1

BetsyMcPhail, 2019-10-07:
The mac unprovisioned builds updated numpy from 1.16.4 to 1.17.1. As a result, [test_linear_algebra ](https://github.com/RobotLocomotion/drake/blob/1970a3a31b1e2c4a4fe9f9d564050c0edd9e6045/bindings/pydrake/test/autodiffutils_test.py#L186) is failing.

Failing builds:
https://drake-cdash.csail.mit.edu/testDetails.php?test=34480669&build=1230814
https://drake-cdash.csail.mit.edu/testDetails.php?test=34480711&build=1230918

test_linear_algebra [verifies that matmul is not supported for dtype=object](https://github.com/RobotLocomotion/drake/blob/1970a3a31b1e2c4a4fe9f9d564050c0edd9e6045/bindings/pydrake/test/autodiffutils_test.py#L194) (numpy issue #11332). 

 As of numpy 17.0.1 , object arrays are supported in matmul (https://github.com/numpy/numpy/pull/13503).

BetsyMcPhail, 2019-10-07:
\cc @EricCousineau-TRI

EricCousineau-TRI, 2019-10-07:
Sorry for delayed direct response, but thanks! Filed the PR.

PR #12153 (2019-10-08): py autodiff, symbolic: Support newer features in NumPy 1.17.0
Resolves #12151

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12153)
closed 2019-10-08 by EricCousineau-TRI