RobotLocomotion/drake #12154: Remove deprecated jacobian methods 2019-10-01

jwnimmer-tri, 2019-10-07:
We have some methods labeled to be removed now-ish, but I do not feel confident to attempt the PR that removes them.

@mitiguy Are you able to open a PR to remove these?  Maybe within the next one week?

```console
./plant/test/kuka_iiwa_model_tests.h:139:  DRAKE_DEPRECATED("2019-10-01", "Use CalcJacobianSpatialVelocity().")
./plant/multibody_plant.h:1579:  DRAKE_DEPRECATED("2019-10-01", "Use CalcJacobianTranslationalVelocity().")
./plant/multibody_plant.h:1686:  DRAKE_DEPRECATED("2019-10-01", "Use CalcJacobianTranslationalVelocity().")
./plant/multibody_plant.h:1748:  DRAKE_DEPRECATED("2019-10-01", "Use CalcJacobianTranslationalVelocity().")
./tree/multibody_tree.cc:1091:// DRAKE_DEPRECATED("2019-10-01", "Use CalcJacobianTranslationalVelocity().")
./tree/multibody_tree.cc:1119:// DRAKE_DEPRECATED("2019-10-01", "Use CalcJacobianTranslationalVelocity().")
./tree/multibody_tree.cc:1235:// DRAKE_DEPRECATED("2019-10-01", "Use CalcJacobianTranslationalVelocity().")
```

mitiguy, 2019-10-07:
@jwnimmer-tri   Yes - I'll do this within a week.

jwnimmer-tri, 2020-01-07:
@mitiguy Is this issue completed now?

mitiguy, 2020-01-18:
This issue is closed.  Issue #12600 has been opened to deprecate method
CalcFrameGeometricJacobianExpressedInWorld()  in favor of  
CalcJacobianSpatialVelocity().

closed 2020-01-18 by mitiguy