RobotLocomotion/drake #12164: Proposal: Add optional "monitor" function to Simulator::AdvanceTo()

sherm1, 2019-10-08:
Currently `AdvanceTo()` takes a single argument, the time at which to return. We have had many requests for being able to terminate due to other events (see #4447) and have a general scheme in mind (and partially implemented in Drake) for doing this from EventHandlers (with some awkward compatibility issues). 

This proposal is a simple, backwards-compatible scheme (inspired by a comment from @jwnimmer-tri) that addresses two long-standing Drake needs:
- It addresses many of the "early termination" needs in #4447, and
- it provides a way to get "Diagram level" trajectories out a simulation (#4114).

The idea is to add an optional second argument to `AdvanceTo()`:
```c++
template <typename T>
AdvanceToResult Simulator<T>::AdvanceTo(
    const T& boundary_time,
    std::function<EventStatus>(const System<T>&, const Context<T>&) monitor = nullptr);
```

The semantics are similar to a per-step Publish event handler, but with much less boilerplate (just a lambda will do fine). If a monitor() is provided, the Simulator calls it after each successfully-completed internal step, supplying the full top-level Context to the monitor, and checks its return status for termination. monitor() can use the context to check for successful or unsuccessful termination, or can publish the context value to obtain a complete record of the trajectory. 

What this does _not_ do is alter the simulation in any way (other than terminating it) -- the monitor cannot affect the step size choices at all, so may not detect a termination or error condition immediately.

One question on the API -- it's not strictly necessary to pass the System along with Context since that could be captured in the functor. But it seemed cleaner this way since the Simulator can guarantee that the (System,Context) pair is legitimate.

Thoughts?

/cc @edrumwri @jwnimmer-tri @RussTedrake @mpetersen94

edrumwri, 2019-10-08:
We do get asked to provide this functionality continually, and having something in place would be better than saying "just call AdvanceTo(.) in small increments until your condition is met".

But passing in the `Context` and the `System` doesn't seem to add anything. If I'm calling the Simulator's AdvanceTo(.) method, I know what System I'm simulating, right? So I couldn't use the callback to determine which of possibly several System's I might be concerned with. And same with the Context. Or am I missing something?

sherm1, 2019-10-08:
The monitor() shouldn't know the current Context value -- we have to pass that in for the same reason we pass it to event handlers (the Simulator is the owner of the Contexts it modifies). We don't pass the System to event handlers because they always belong to some sub-System so presumably have access to it. My thought is that the monitor() is more global so unlikely to belong to the System it is monitoring.

jwnimmer-tri, 2019-10-08:
I like the general proposal for a `monitor` functor.

As to the signature, we should evaluate potential signatures by writing example calling code and show how the signature operates for those acute uses.  Speculation in the absence of examples is a difficult way to articulate the trade-offs of API choices.

I am a little worried about reusing `class EventStatus` as the return type.  I would be more comfortable with a functor-specific data type, even if its implicitly convertible from EventStatus.  I don't thinks like "The state was updated successfully" make sense in the sense of this handler, and if we ever want to either add more complexity to the EventStatus for actual events, or eschew complexity for monitor functors, we will be stuck.

sherm1, 2019-10-08:
Regarding timing of monitor() calls: since monitor() has Publish semantics it should be called at the points where publishing can occur -- that is, at the "t-" locations at the end of initialization and at the end of each step.

sherm1, 2019-10-08:
@jwnimmer-tri since EventStatus is already used for Publish events, I think it makes sense for monitor() also due to the parallel semantics. They actually are the same thing (a per-step Publish event) and would be handled identically by the Simulator. For convenience, I would propose an additional overload that accepts a void-returning monitor() that is presumed always to succeed.

jwnimmer-tri, 2019-10-08:
Okay, I am content to argue about `EventStatus` with specific code on the table.  Maybe it will be fine.

edrumwri, 2019-10-08:
> The monitor() shouldn't know the current Context value -- we have to pass that in for the same reason we pass it to event handlers (the Simulator is the owner of the Contexts it modifies). We don't [...]

I'm imagining that the code could look like:

    Simulator s(my_system);
    std::function<bool()> monitor = [&my_system, &s.get_context()] { // Check something here. }
    s.AdvanceTo(some_time, monitor);

So I'm still wondering what the API you've proposed buys.

sherm1, 2019-10-08:
@edrumwri you mean what is the advantage of returning EventStatus rather than bool? It allows for error returns with a message (using the existing EventStatus class). That's the same as any other event handler can return and can be very helpful over silent termination.

edrumwri, 2019-10-08:
Sorry, red herring: please assume that my functional returns `EventStatus` and otherwise read my question the same way.

sherm1, 2019-10-08:
Oh, gotcha. I edited the original comment and deleted the spurious comments.

I don't mind the `&my_system` capture too much, but the `&s.get_context()` implies that the Simulator promises to have updated THAT context by the time of the monitor call. AFAIK we make no such promise and the Simulator always passes a context to anyone who cares. I don't think we should make that promise -- the simulator should be able to construct and maintain multiple contexts (say for future threaded work, or efficiency improvements). Without the context it would be harder to draw the analogy between the monitor() and a Publish handler.

The reason I'm not totally wild about the `&my_system` capture is just that there is now no guarantee that the System is the one actually being simulated. The Simulator is in a great position to be able to pass in the true System (and true trajectory Context). But I could live without the System if you hate it, and that would make the monitor() signature identical to a publish handler's.

edrumwri, 2019-10-08:
I don't hate passing in a System, just didn't see much use for it.

And I agree with passing in the `Context`. A monitor would always seem to need a Context.

jwnimmer-tri, 2019-10-08:
I would like to see the example under study include subsystems (i.e., use a Diagram).  Digging the subcontext out of the main context is no fun.  That could affect the trade-offs of what is gets captured vs what is an argument.

sherm1, 2019-10-08:
That's a good point. My thought was that the monitor() should operate at the same level of abstraction as the Simulator -- that is, with just the top-level System visible, since we already have the ability to supply subsystem-level handlers. But it seems very likely that people would use this to have some low-level subsystem check something that they didn't export to the top level. Better to get out in front of that.

edrumwri, 2019-10-08:
> I would like to see the example under study include subsystems (i.e., use a Diagram). Digging the subcontext out of the main context is no fun. That could affect the trade-offs of what gets captured vs what is an argument.

+1

sherm1, 2019-10-18:
OK, here is my current shot at an API for the Simulator monitor() function. Rather than supply the function on every call, it is set with a `simulator.set_monitor()` call. Here is an example showing how a monitor can be written that digs down to a subsystem to check for termination. I hacked the cart_pole example to terminate when the internal plant reaches a certain angle:
```c++
  // cart_pole is defined earlier and used to build this diagram.
  systems::Simulator<double> simulator(*diagram, std::move(diagram_context));

  simulator.set_monitor([&cart_pole](const systems::System<double>& system,
                                     const systems::Context<double>& context) {
    const RevoluteJoint<double>& pin_joint =
        cart_pole.GetJointByName<RevoluteJoint>("PolePin");

    const systems::Context<double>& subcontext =
        system.GetSubsystemContext(cart_pole, context);
    const double angle = pin_joint.get_angle(subcontext);
    if (angle < -1.5) {
      return systems::EventStatus::ReachedTermination(
          &cart_pole, fmt::format("Reached target angle {}.", angle));
    }
    return systems::EventStatus::Succeeded();
  });

  auto status = simulator.AdvanceTo(10.);
  std::cout << status.FormatMessage() << std::endl;
```
This runs and produces the following output:
```
Simulator returned early at time 0.698685 because MultibodyPlant<double> System
'::cart_pole_diagram::cart_pole_plant' requested termination with message: 
"Reached target angle -1.50377."
```

Note that the Simulator cannot provide a Diagram or DiagramContext since it works on any System. To make the above work without ugly dynamic_casts I moved GetSubsystemContext() up from Diagram to System (it worked unchanged there since the underlying virtual was already defined in System). Personally I think that's an improvement, but I suspect you will have an objection @jwnimmer-tri. But as you pointed out access to subsystems is likely to be very common and I do not want to be adding any more APIs that force users to do dynamics_casts. An alternative would be _not_ to pass the System at all; the user would have to capture the diagram as well as the subsystem. That introduces the possibility that the wrong System is captured -- I prefer the security of having the Simulator pass in exactly the System that matches the passed-in Context. (And we must pass the context as an argument because we don't---and shouldn't---promise that the user can know which context object will be the one that is provided for any particular step.) 

Thoughts, please?

jwnimmer-tri, 2019-10-21:
> ... we must pass the context as an argument because we don't---and shouldn't---promise that the user can know which context object will be the one that is provided for any particular step ...

Why is this true?  I agree that for event handlers defined _within_ a System, we need to pass the context.  But this is an _extrinsic_ monitor, and is always and only to be called after a committed step -- not a putative step.  If the user were achieving this effect by calling `AdvanceTo(now + dt)` instead of the new sugar, they would definitely be inspecting `simulator.get_context()` to see the current state.

A no-argument callback seems substantially simpler to me:
```c++
  systems::Simulator<double> simulator(*diagram, std::move(diagram_context));

  simulator.set_monitor([&cart_pole, &cart_pole_context]() {
    const RevoluteJoint<double>& pin_joint =
        cart_pole.GetJointByName<RevoluteJoint>("PolePin");
    const double angle = pin_joint.get_angle(cart_pole_context);
    if (angle < -1.5) {
      return systems::EventStatus::ReachedTermination(
          &cart_pole, fmt::format("Reached target angle {}.", angle));
    }
    return systems::EventStatus::Succeeded();
  });

  auto status = simulator.AdvanceTo(10.);
  std::cout << status.FormatMessage() << std::endl;
```

The user will often already have a reference to the subsystems and subcontexts in scope of their main (or if not, getting it is no harder outside of the lambda than inside it).

sherm1, 2019-10-21:
> Why is this true? I agree that for event handlers defined within a System, we need to pass the context. But this is an extrinsic monitor, and is always and only to be called after a committed step [...]

The monitor is semantically identical to a Per-step Publish handler -- it is called at committed substeps, which is not the same as a return from AdvanceTo() (to make that distinction is why we changed that method's name). We do promise to have the trajectory represented in `simulator.get_context()` after a return from AdvanceTo(), but to date we have not yet made such a promise for substeps. The reason I'm reluctant to do that in general is that for speed we may want to use interpolation to allow integrators to take bigger steps while still delivering more-frequent substeps, and having a second Context can be a good way to do that. Simbody uses that method and I don't want to preclude Drake from doing the same.

However, there is no similar argument to be made for passing in the System. I like that, but only mildly and would be fine with requiring the monitor user to capture it. Having just the context would also strengthen the parallel to a per-step Publish handler which also gets only the context.

edrumwri, 2019-10-21:
This was my prescription far earlier too, but IMO Jeremy's does not look simpler (he's leaving out the part where `cart_pole_context` is obtained). Jeremy's does appear more efficient.

I would have to see more, but my initial feeling is that I wouldn't like calling a monitor with a hidden Context. I could see someone doing the binding Jeremy did and being unaware that the two Contexts are not the same.

jwnimmer-tri, 2019-10-21:
Doesn't that get down to the resolution at which the monitor is queried?  Maybe we could loosen that to allow for the better ergonomics?

For myself at least, I will not use the monitor API you proposed.  Writing out the context junk is just too much typing, and will be worse the more subsystems I need to talk to.

jwnimmer-tri, 2019-10-21:
> ... he's leaving out the part where cart_pole_context is obtained.

I did omit it from my paste, but I did not have to write it anew.  The main program already had a variable with the correct context, I just had to capture it in the lambda.  My thesis is that main programs will almost always already have the variable they need.

sherm1, 2019-10-21:
I don't think the (possible) convenience of grabbing an existing subcontext is worth the architectural tradeoff of precluding use of an interpolated context later. I will PR the exactly-parallel-to-per-step-Publish version (that is, just the full-context argument) with an example or two and we can discuss further there.

sherm1, 2019-10-22:
@jwnimmer-tri here is the latest attempt:
```c++
  simulator.set_monitor([&cart_pole,
                         &pole_pin](const systems::Context<double>& context) {
    const double angle = pole_pin.get_angle(cart_pole.FindMyContext(context));

    if (angle < -1.5) {
      return systems::EventStatus::ReachedTermination(
          &cart_pole, fmt::format("Reached target angle {}.", angle));
    }
    return systems::EventStatus::Succeeded();
  });
```

FindMyContext() lets a subsystem extract its subcontext given access to the root context (that's done through the existing "parent service" interface).

PR #12213 (2019-11-05): Adds a monitor function for Simulator trajectories
Adds an optional monitor() function to a Simulator() that can be used to publish trajectories, check for termination, or report errors. Also adds a status return from Simulator::AdvanceTo() and related methods so that the monitor() results can be reported (backwards compatible if ignored).

The monitor() function is semantically identical to a top-level, per-step Publish event handler. It uses the pre-existing EventStatus class to report success, termination, or failure.

Resolves #12164
Resolves #4114

---
This change is [[image]](https://rev [... cut]
closed 2019-11-05 by sherm1