RobotLocomotion/drake #12173: Needed functions for bitwise equality.

edrumwri, 2019-10-09:
Unit tests for outputting hydroelastic contact information in MBP (see #12162) are currently using customized methods for validating bitwise equality of SurfaceMesh, ContactSurface, and MeshField types. @sherm1 pointed out that putting these functions into those classes would improve maintainability, and I agree, but don't want to hold off on merging #12162 since implementing, documenting, and testing each of those `Equal()` functions will take some work.

TLDR; move Equal(.) functions in `multibody_plant_hydroelastic_contact_results_output_test.cc` to individual classes.

Assigning to Damrong and Bel- please feel free to reassign.

antequ, 2019-10-15:
I don't like how we call it "bitwise equality" in this issue. Can we use a better term, like "exact equality" or "deep equality" or "deep, exact equality"?

Bitwise equality implies converting floats to longs and comparing them, as in the second code example given here: https://docs.nvidia.com/cuda/cuda-c-programming-guide/#atomic-functions (so that NaNs equal NaNs and don't cause an infinite loop).

sherm1, 2019-10-15:
Per f2f with Bel and Ante: Fine to fix this for now by documenting the existing behavior. We could call this "deep, exact equality" but with IEEE standard NaN treatment -- that is, if there is a NaN anywhere the objects will test "not equal".

For unit test purposes it would be useful to have the equality tests consider NaN==NaN to be true. But we don't have a use case for that at the moment so it would be fine just to add a TODO suggesting a possible improvement, along with documentation of the current behavior.

PR #12193 (2019-10-21): Move equal functions from test into their respective classes
Closes #12173

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12193)
closed 2019-10-21 by tehbelinda