RobotLocomotion/drake #12191: Python bindings: Change CalcFrameGeometricJacobianExpressedInWorld -> CalcJacobianSpatialVelocity

mitiguy, 2019-10-14:
As part of issue #10155, the term "geometric Jacobian" is deprecated in favor of 
"Jacobian with respect to generalized velocities v".
Jacobian notation has been updated accordingly.  See:
https://drake.mit.edu/doxygen_cxx/group__multibody__quantities.html

It would be ideal for someone (e.g., Eric C) who is highly familiar with Python and the Drake Python bindings to make this change appropriately in the Python bindings. 

CalcFrameGeometricJacobianExpressedInWorld() is the remaining method in Drake (other than in the attic) that is called ....GeometricJacobian....

EricCousineau-TRI, 2019-10-15:
I'd be fine with doing this if it were only touching the Python parts. Do you have a draft that resolves the TODO in the C++ code, and then breaks the Python code?
https://github.com/RobotLocomotion/drake/blob/master/multibody/plant/multibody_plant.h#L1634-L1635
If so, then I can go and fix the Python parts.

FTR, deprecation is simply pattern matching the examples enumerated here, once you've deprecated the C++ code:
https://github.com/RobotLocomotion/drake/blob/dd9442172c6458b9722b9beda9e3c54d6b50acaa/bindings/pydrake/common/test/deprecation_example/cc_module_py.cc#L50-L66

EricCousineau-TRI, 2019-10-15:
Ah, Python is the only usage. I'll do this, then. Thanks!