RobotLocomotion/drake #12196: InitialValueProblem for backward integration

akshitj1, 2019-10-15:
Hi, I was trying to implement time varying LQR for which I need backward integration. This is required for finding S, where we have a ode for S and its value at t_final [[referring from here](http://underactuated.csail.mit.edu/underactuated.html?chapter=lqr)] . To do this, I was trying to use InitialValueProblem solver in drake. But it does not allows final time to be lesser than initial time [[ref](https://github.com/RobotLocomotion/drake/blob/5855426f7be7055e2ec7a0c0b6a7beac315ccfc4/systems/analysis/initial_value_problem-inl.h#L233)] . So, I tried to use negative times so that, my initial time is lesser than final time but then, negative times are also not allowed [[ref](https://github.com/RobotLocomotion/drake/blob/20c3209a76d1553f17e65d882d0df4c3ad05c53e/systems/analysis/integrator_base.h#L418)]. Am I doing something wrong or there is some other solver to do backward integration. Any help would be great.

edrumwri, 2019-10-15:
Hi @akshitj1, negate the sign on your derivatives so that you can integrate forward in time.

akshitj1, 2019-10-16:
Oh yes, that should work. Thanks @edrumwri

closed 2019-10-16 by akshitj1