RobotLocomotion/drake #12204: procman sheriff: Needs porting to Python 3

EricCousineau-TRI, 2019-10-17:
We use procman sheriff (and its GUI) in Anzu, and we're planning to remove Python 2 support.

Per f2f with Jamie, this tool has Python 2-only libs (with no Python 3 ports), so it would involve more than just 2to3 updates.

\cc @jwnimmer-tri

jwnimmer-tri, 2019-10-17:
Given that the tool's use of python2 is unrelated to our own source code (no cross-calling), and we use a pre-compiled version of procman, I'm not sure that we should care about this.  We're using 18.04 and it has python2 available.  I guess the question is when we switch Anzu to 20.04 and whether procman still works on that platform.  If it does, I don't see any need to mess with something that's not broken.

EricCousineau-TRI, 2019-10-17:
Sounds good. We'll still look at the dependencies, estimate how long it'd take; if it's non trivial (more than 2 calendar days), then we punt.

BetsyMcPhail, 2019-11-07:
First try of updating from GTK2 to GTK3: https://github.com/RobotLocomotion/libbot2/pull/20

EricCousineau-TRI, 2019-11-21:
Here's an example PMD from the past:
https://github.com/RobotLocomotion/drake/blob/5c7d53403e2d976ac8a46c466ab1a5bb410d2107/examples/kuka_iiwa_arm/dev/box_rotation/iiwa_dual_box_rot.pmd
(Just for an example.)

EricCousineau-TRI, 2019-12-05:
Will try this out with Anzu when it's ready.

Retrospective: stayed within 2 day's worth of work. Mainly just renaming.

EricCousineau-TRI, 2019-12-12:
Just need to make the packages.

EricCousineau-TRI, 2020-01-30:
Last item here is to fix monospace for output console, then should be good to try it out in Anzu.

EricCousineau-TRI, 2020-03-19:
Sorry for not posting earlier, but I had tried this out on our hardware setup on 2020-02-28. Worked fine!

@jamiesnape Can we move forward on getting these packages up?

jamiesnape, 2020-04-22:
Version 0.0.1.20200422-1 is available on drake-apt.csail.mit.edu/bionic bionic main.

closed 2020-04-22 by jamiesnape