RobotLocomotion/drake #12207: Mass matrix calculation in MultibodyPlant is too slow

sherm1, 2019-10-17:
In an O(n) multibody code it is preferable to treat the mass matrix and its inverse as operators M(⋅) and M⁻¹(⋅) that produce a matrix-vector product in O(n) time. However, many robotics users prefer to obtain the O(n²) mass matrix (a.k.a. "inertia matrix") explicitly. Currently MutibodyPlant provides that via [`CalcMassMatrixViaInverseDynamics()`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#ac3238273c2c927101e4515147bed17f6) which essentially makes _n_ calls to the M(⋅) operator applied to the columns of an identity matrix. That produces a correct result but, as measured by @mposa in #10322, is inefficient compared to a bespoke method that knows it is constructing a full mass matrix. 

We should provide a method `MultibodyPlant::CalcMassMatrix()` that forms and returns a mass matrix as efficiently as possible using, for example, Abhi Jain's composite-body inertia algorithm 4.2 (pg. 64 in _Robot and Multibody Dynamics_). That algorithm computes the mass matrix directly and takes advantage of its symmetry also. Please be sure this can be AutoDiffed.

The new method can be conveniently unit tested against the current method since they compute the same matrix using entirely distinct algorithms!

Tentatively assigning to @mitiguy, with cc @amcastro-tri as the ideal consultant and feature reviewer.

edrumwri, 2019-10-17:
Even though there is a preference for the mass matrix, perhaps leaving it as slow would be preferable barring an influx of resources. All of the applications that I have in mind in robotics would be better done using the operators that @sherm1 cited (and then we have to re-educate users).

Regardless, I would downgrade the priority for this.

sherm1, 2019-10-17:
I agree with @edrumwri that people should use operators rather than an explicit mass matrix. However I still think this is a high priority due to
- current practice and a limited ability on our part to retrain outside users,
- a Principle of Least Astonishment corollary that says if we offer a method to calculate the mass matrix we should do a good job of it (at least as well as other robotics sim codes do!), and
- an unfortunate and surprising lack of _actual_ operator methods in MultibodyPlant! (See #11389 and #12069 for example), and
- a documented performance regression from RigidBodyPlant (#10322) that is one of the items preventing us for removing that obsolete code from Drake.

amcastro-tri, 2019-10-22:
I'd like to request that this method returns an `Eigen::SparseMatrix`, given MBP knows the sparsity pattern.

sherm1, 2019-10-22:
That's a great idea (sparse mass matrix)! I think that adds an additional layer of complexity though -- maybe start with the dense matrix and treat sparse as a 2nd project? It would be easy just to return just half the matrix since it's symmetric (and Eigen has methods for dealing with that). But I'm not sure how much that saves over filling in the whole thing.

edrumwri, 2019-10-22:
Perhaps use the type of matrix as an output argument? For a relatively small system, I'd expect manipulating a dense matrix to be much faster than a sparse one.

sherm1, 2020-03-11:
I believe this is fixed now by #12764 and #12809. It's not returning a sparse matrix but `CalcMassMatrix()` should be _much_ faster than `CalcMassMatrixByInverseDynamics()`.

@mposa please let us know whether you are satisfied with this and open a new issue if not.

closed 2020-03-11 by sherm1