RobotLocomotion/drake #12210: MultibodyPlant is missing basic operators

sherm1, 2019-10-17:
Best practice in an O(n) multibody code is to use O(n) operators rather than O(n²) explicit mass and Jacobian matrices -- or worse, O(n³) factorizations. As discussed in #12207, some retraining of users is needed to get them to think in terms of operators. That is hard enough under the best of circumstances, but is made particularly awkward by the fact that we don't currently have those operators explicitly available in the MultibodyPlant API!

We need to implement the following operator methods:
- [ ] `MultiplyByMassMatrix()` f = M a
- [ ] `MultiplyByMassMatrixInverse()` a = M⁻¹ f
- [ ] `MultiplyBySystemJacobian()` V = J v (spatial velocity from generalized velocity)
- [ ] `MultiplyBySystemJacobianTranspose()` f = Jᵀ F (generalized force from spatial force -- issue #12069)

avalenzu, 2019-10-17:
`SystemJacobian` is not an "Approved Jacobian Name"™ per the Treaty of  #10155. Lest we risk re-opening the Jacobian naming war, we should probably follow the conventions outlined there. ;-)

sherm1, 2019-10-17:
Thanks for the reminder @avalenzu -- I wouldn't want to reopen that battle! I don't see any name that looks appropriate for the basic operators that appear in the equations of motion though -- did we come up with names for those? I'm thinking of the J that appears in `M v̇ = τ + JᵀF`.

sherm1, 2019-10-17:
Uh oh -- lots of discussion here on this name! I think the best-conforming names would be something like:
- `MultiplyJacobianSpatialVelocityWrtV()`
- `MultiplyJacobianSpatialVelocityWrtVTranspose()

But even those don't capture the fact that the system Jacobian always works with the full set of bodies (using the body-origin spatial velocity of each body). Are those too obscure? If so we could define "SystemJacobian" to mean "full size SpatialVelocity Jacobian with respect to v" and provide shorter names.

cc @mitiguy @edrumwri

edrumwri, 2019-10-17:
My personal preference would be to just use `MultiplyJacobianSpatialVelocityWrtV`-type terminology and leave it to the argument type (`const std::vector<SpatialVelocity>&` and `const std::vector<SpatialForce>&`, respectively) and documentation to disambiguate from the methods that return the matrix.

avalenzu, 2019-10-17:
My 2 cents: 

## For a single set of frames A, B, and E
We should have a `MultiplyJacobianSpatialVelocity()` method such that 
```
V = MultiplyJacobianSpatialVelocity(..., s);
``` 
gives the same result as
```
CalcJacobianSpatialVelocity(..., &Js_V); 
V = Js_v * s;
```
where `V` is a single spatial velocity, `s` is v or qdot (as in the doc for `CalcJacobianSpatialVelocity()`), and "..." are all the other inputs to `CalcJacobianSpatialVelocity`.

We could also have an analogous `MultiplyJacobianSpatialVelocityTranspose()`. 

## For the Jacobian of all the body-frame spatial velocities
I'd propose `MultiplyJacobianFoo()` where "Foo" is an as yet undecided term. @sherm1 has proposed "System," but I would argue that that's a heavily overloaded term in Drake already and as such may not be the best choice.

avalenzu, 2019-10-17:
Also, I think we should avoid decorating API method names with `WrtV` (helpers in tests are fine). We decided (after substantial back and forth) on using the `JacobianWrtVariable` enum for distinguishing between the two cases in the `CalcJacobianFoo()` methods. Absent a compelling reason, we should be consistent here.

sherm1, 2019-10-17:
Hmmm ... I see your point about SystemJacobian since we have a rather important class in Drake called System! Maybe _Full_ Jacobian?

sherm1, 2019-10-17:
BTW these operators should not be complicated with lots of parameters, frames, etc. -- they should just be the equivalent of the matrices that appear in the pedagogical equations of motion. That way they are easy to use and understand and can run at maximum speed.

sherm1, 2019-10-17:
Also it doesn't make sense to have a wrt parameter -- writing the equations of motion in terms of qdot would mean (in general) that the nq x nq mass matrix is not invertible.

avalenzu, 2019-10-17:
Re. lots of parameters: Sorry I wasn't clear above: the frames etc. were only part of my proposal for an operator version of the current `CalcJacobianSpatialVelocity()`. I think we can safely disregard that for now if it's out of scope for this issue.

Re. Full Jacobian: How about `MultiplyByPlantJacobian()`?

Re. Wrt: If it only makes sense to have the to-be-named Jacobian be wrt v, then I think we'd do best to forgo both the wrt parameter _and_ the `WrtV` suffix.

sherm1, 2019-10-17:
Oooh ... I like "PlantJacobian"! And I agree we should just drop the whole WrtV stuff and simply document that a "Plant Jacobian" is always wrt v.

Anyone object to PlantJacobian or have a better idea?

mitiguy, 2019-10-17:
I like the suggestions, with a small change to "MultiplyBy".
I expect a method named MultiplyByMatrixJoe() to return   
something * Joe,   not   
Joe * something.

Some alternatives:
`MultiplyPlantJacobian()` 
`MultiplyPlantJacobianBy()` 
`MultibodlyPlantSpatialJacobian()`
`MultibodlyPlantSpatialJacobianBy()`

Similarly for the transpose, e.g.,  
`MultiplyPlantJacobianTranspose()`

avalenzu, 2019-10-18:
Interesting, @mitiguy. I actually have a strong expectation for the opposite behavior. For me "multiply x by A" means "A * x." I'd be interested to know other people's take on this.

sherm1, 2019-10-18:
I have the same sense of that as you Andres, but Paul says it means the opposite to him. I didn't realize before that "By" was ambiguous, but since apparently it is we can stay out of trouble if we leave it out of the name.

edrumwri, 2019-10-18:
I'm with Paul on this one. Multiply "x by A" means "x * A" to me.

sherm1, 2019-10-18:
@edrumwri the use here would be `MultiplyByPlantJacobian(x)`. Does that still have the backwards sense for you?

edrumwri, 2019-10-18:
@sherm1 It does seem backward.