RobotLocomotion/drake #12251: py mbp: Permit overriding ForceElement?

EricCousineau-TRI, 2019-10-28:
See this Stack Overflow post: https://stackoverflow.com/questions/58572398/calling-super-init-on-subclass-of-forceelement-causes-no-constructor-def

EricCousineau-TRI, 2019-10-28:
Hm... It looks like `ForceElement` is heavily dependent on internal classes, like `MultibodyTree`, `PositionKinematicsCache`, `VelocityKinematicsCache`, etc., :(

Some potential solutions:
(a) Bind the minimum set of internal classes. However, this is not great from a user or developer perspective, because they should be able to evolve without any care for public API.
(b) Make "trampoline" force element API in C++, which only uses public classes (`MultibodyPlant`, `Context`, etc.), hopefully in such a way that avoids infinite recursion or invalid calculations (e.g. on dirty or stale data).
(c) Make "trampoline" force element API in Python only. Pretty much the same as (b), but only for Python bindings. This would probably be 100x more painful to develop.

Really, I like (b) and (c) because it promotes user-defined `ForceElement`s without having to dip into internal API.

@amcastro-tri @sherm1 Have y'all given this any thought before?

amcastro-tri, 2019-10-28:
For the particular problem stated in [Stack Overflow ](https://stackoverflow.com/questions/58572398/calling-super-init-on-subclass-of-forceelement-causes-no-constructor-def) I believe a `ForceElement` is not the proper solution. I replied in that thread.

closed 2024-03-21 by jwnimmer-tri