RobotLocomotion/drake #12252: Docs about model instances.

amcastro-tri, 2019-10-28:
We have detailed docs about model instances [here](https://github.com/RobotLocomotion/drake/blob/3c9ef6d3ae39200b0e36d737ccb80f18554b816a/multibody/tree/model_instance.h#L18). However they are not easy to discover given that they belong to an internal implementation. 

Probably we could move (or even copy) the publicly relevant sections of those docs to the `MultibodyPlant` class itself? Would that be a good place? or do people have other suggestions on where to place this information to make it more visible?

cc'ing @edrumwri, who pinged me about this.
cci'ing @sammy-tri, the author of most of our model-instance stuff.
cc'ing @EricCousineau-TRI, @siyuanfeng-tri, @sherm1, @avalenzu for opinions.

edrumwri, 2019-10-28:
I'd suggest having a model instances doxygen group in MBP (could include `num_model_instances()`, `AddModelInstance()`, etc.) and including this info there.

EricCousineau-TRI, 2019-10-28:
SGTM. Another option is to keep those docs in `multibody/tree/model_instance.h` and to cross-reference this doc section from the model instance doxygen group. However, Evan's suggestion makes things easier to discover.

sherm1, 2019-10-28:
We also have issue #11807 about cleaning up MBPlant doxygen in general.

edrumwri, 2019-11-08:
This has been resolved.

closed 2019-11-08 by edrumwri