RobotLocomotion/drake #12257: VectorBase output with operator<<() injects unwanted punctuation

sherm1, 2019-10-28:
Testing the new `simulator.monitor(context)` feature I tried this:
```c++
std::cout << context.get_time() << " " << context.get_continuous_state_vector() << "\n";
```
A typical output line was:
```
0.616667 [0.0814282, -1.08041, 0.127596, -5.96037]
```
The brackets and commas make this awkward to process further. In contrast, if I use
```c++
auto eigen_vector = context.get_continuous_state_vector().CopyToVector();
std::cout << context.get_time() << " " << eigen_vector.transpose() << "\n";
```
I get
```
0.616667 0.0814282  -1.08041  0.127596  -5.96037
```
with no unwanted fluff. That can be directly piped to a plotter, read in to Matlab, etc.

I would like to be able to dump out the state using the simpler code above but have it come out with just numbers and no syntax. Would anyone object to my removing the brackets and commas from `operator<<(VectorBase)`? That could potentially break some code somewhere if anyone is expecting brackets and commas, but I think the plain output format is more likely to be useful.

avalenzu, 2020-05-20:
Should we output an empty string for an empty vector? I could see that being easy to miss in console output.

avalenzu, 2020-05-20:
An alternative would be to keep the empty vector output unchanged ("[]"), while removing the punctuation from non-empty vectors.

sherm1, 2020-05-20:
The problem I wanted to resolve was the inconsistency between Eigen vectors and VectorBase vectors. If they behave the same way we can mix and match them in output. I _think_ Eigen vectors never output punctuation. So a user can write `std::cout << "[" << v << "]\n";` and get the same output without knowing whether v is and Eigen or VectorBase vector. And if both simply print nothing for a 0-length vector they will get `[]` as you proposed.

avalenzu, 2020-05-20:
Got it. I'll update the PR.

EricCousineau-TRI, 2020-05-22:
@sherm1 I don't really understand your workflow. Why are comments an antithesis to your workflow? Or rather, why are you *losing* structure for being able to paste things to MATLAB?

FTR, I don't love the new string formatting in Python in #13365; it's `__repr__` is no longer code that can be copy+pasted.

If you want a text script that can help you process it, it's like ~3 lines of regex in Python. I don't think shimming in a lack of structure into `operator<<` is the correct answer?
(Seeing this now due to Anzu mention from @jwnimmer-tri )

jwnimmer-tri, 2020-05-22:
I'm ambivalent on what the C++ ostr format should be, but for pydrake we definitely want the repr-like string format to remain (with brackets and commas).

EricCousineau-TRI, 2020-05-22:
Specifically:
https://docs.python.org/3.4/library/functions.html#repr
> [image]

EricCousineau-TRI, 2020-05-22:
> Testing the new `simulator.monitor(context)` feature I tried this: [...]

Just to check, is there an open issue on this feature, or is it still in the works! Sounds great!

And for this workflow (printouts that are MATLAB-friendly), I believe there should be more site-specific customizations for this, rather than de-structuring the output in all of Drake.

jwnimmer-tri, 2020-05-22:
The monitor is [Simulator::set_monitor](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_simulator.html#aa92bce4d4205f51b3b17bcaae09566f7).

EricCousineau-TRI, 2020-05-22:
TIL, thanks!

jwnimmer-tri, 2020-05-22:
If nothing else, the current pydrake unit test was not clear enough to Drake maintainers that the exact string output under test was important as a golden answer, and should not be changed (at least not without consulting the SMEs).  We will need to improve the clarity of the test.

EricCousineau-TRI, 2020-05-22:
Duly noted. The intricacies of adhering to `__repr__` should also somehow be made common knowledge.

sherm1, 2020-06-02:
Fixed by #13365 (reverted, then resurrected as #13464).
Thanks Andrés and Eric!

PR #13365 (2020-05-22): systems: Use Eigen's formatting in VectorBase::operator<<() output
Resolves #12257.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13365)
closed 2020-06-02 by sherm1