RobotLocomotion/drake #12283: Deprecation timeline for less popular attic items

jwnimmer-tri, 2019-10-30:
In #12158, we are discussing the timeline for deprecating and removing RigidBodyTree.  That is the most widely used item in the attic, and so will stay around the longest (be the last attic item to be removed).

In the meantime, I think that some of the less widely used items in the attic could be removed earlier, to improve build times and make progress towards the attic-less goal.

I'll post some proposals below.

Relates to #10079.

jwnimmer-tri, 2019-10-30:
Here's a concrete proposal for a first pass.

Leave these items alone:
- [ ] attic/manipulation/planner/**
- [ ] attic/multibody/**
- [ ] attic/systems/rendering/**
- [ ] attic/systems/sensors/**
- [ ] attic/systems/trajectory_optimization/**
- [ ] attic/util/**
- [ ] attic/systems/controllers/BUILD.bazel
- [ ] attic/systems/controllers/rbt_inverse_dynamics.cc
- [ ] attic/systems/controllers/rbt_inverse_dynamics.h
- [ ] attic/systems/controllers/rbt_inverse_dynamics_controller.cc
- [ ] attic/systems/controllers/rbt_inverse_dynamics_controller.h
- [ ] attic/systems/controllers/test/rbt_compute_torque.h
- [ ] attic/systems/controllers/test/rbt_inverse_dynamics_controller_test.cc
- [ ] attic/systems/controllers/test/rbt_inverse_dynamics_test.cc

Deprecate these immediately, to be removed in three months:
- [x] attic/manipulation/dev/BUILD.bazel
- [x] attic/manipulation/dev/box_as_table.sdf
- [x] attic/manipulation/dev/double_dumbbell_for_pick_up.sdf
- [x] attic/manipulation/dev/dumbbell_for_pick_up.sdf
- [x] attic/manipulation/dev/quasistatic_iiwa_pick_and_place_demo.cc
- [x] attic/manipulation/dev/quasistatic_system.cc
- [x] attic/manipulation/dev/quasistatic_system.h
- [x] attic/manipulation/dev/remote_tree_viewer_wrapper.cc
- [x] attic/manipulation/dev/remote_tree_viewer_wrapper.h
- [x] attic/manipulation/dev/rod2d.urdf
- [x] attic/manipulation/dev/simplified_schunk_gripper.sdf
- [x] attic/manipulation/dev/test/remote_tree_viewer_wrapper_demo.cc
- [x] attic/manipulation/dev/test/rod2d_time_stepping.cc
- [x] attic/manipulation/models/BUILD.bazel
- [x] attic/manipulation/models/test/dual_iiwa14_polytope_collision_test.cc
- [x] attic/manipulation/models/test/jaco_arm_test.cc
- [x] attic/manipulation/models/test/wsg_50_mesh_collision_test.cc
- [x] attic/manipulation/scene_generation/BUILD.bazel
- [x] attic/manipulation/scene_generation/random_clutter_demo.cc
- [x] attic/manipulation/scene_generation/random_clutter_generator.cc
- [x] attic/manipulation/scene_generation/random_clutter_generator.h
- [x] attic/manipulation/scene_generation/simulate_plant_to_rest.cc
- [x] attic/manipulation/scene_generation/simulate_plant_to_rest.h
- [x] attic/manipulation/scene_generation/test/test_random_clutter_to_rest.cc
- [x] attic/manipulation/sensors/BUILD.bazel
- [x] attic/manipulation/sensors/xtion.cc
- [x] attic/manipulation/sensors/xtion.h
- [x] attic/manipulation/util/BUILD.bazel
- [x] attic/manipulation/util/frame_pose_tracker.cc
- [x] attic/manipulation/util/frame_pose_tracker.h
- [x] attic/manipulation/util/robot_state_msg_translator.cc
- [x] attic/manipulation/util/robot_state_msg_translator.h
- [x] attic/manipulation/util/sim_diagram_builder.cc
- [x] attic/manipulation/util/sim_diagram_builder.h
- [x] attic/manipulation/util/simple_tree_visualizer.cc
- [x] attic/manipulation/util/simple_tree_visualizer.h
- [x] attic/manipulation/util/simple_tree_visualizer_demo.cc
- [x] attic/manipulation/util/test/frame_pose_tracker_test.cc
- [x] attic/manipulation/util/test/robot_state_msg_translator_test.cc
- [x] attic/manipulation/util/test/sim_diagram_builder_test.cc
- [x] attic/manipulation/util/world_sim_tree_builder.cc
- [x] attic/manipulation/util/world_sim_tree_builder.h
- [x] attic/perception/BUILD.bazel
- [x] attic/perception/estimators/dev/BUILD.bazel
- [x] attic/perception/estimators/dev/articulated_icp.cc
- [x] attic/perception/estimators/dev/articulated_icp.h
- [x] attic/perception/estimators/dev/pose_closed_form.cc
- [x] attic/perception/estimators/dev/pose_closed_form.h
- [x] attic/perception/estimators/dev/scene.cc
- [x] attic/perception/estimators/dev/scene.h
- [x] attic/perception/estimators/dev/test/README.md
[... 19 lines cut ...]
- [x] attic/systems/controllers/controlUtil.h
- [x] attic/systems/controllers/polynomial_encode_decode.cc
- [x] attic/systems/controllers/polynomial_encode_decode.h
- [x] attic/systems/controllers/side.cc
- [x] attic/systems/controllers/side.h
- [x] attic/systems/controllers/test/polynomial_encode_decode_test.cc
- [x] attic/systems/controllers/yaml_util.cc
- [x] attic/systems/controllers/yaml_util.h
- [x] attic/systems/controllers/zmpUtil.cpp
- [x] attic/systems/controllers/zmpUtil.h
- [x] attic/systems/estimators/dev/BUILD.bazel
- [x] attic/systems/estimators/dev/pose_estimation_test.cc
- [x] attic/systems/estimators/dev/rotation.h
- [x] attic/systems/robotInterfaces/BUILD.bazel
- [x] attic/systems/robotInterfaces/BodyMotionData.cpp
- [x] attic/systems/robotInterfaces/BodyMotionData.h
- [x] attic/systems/robotInterfaces/QPLocomotionPlan.cpp
- [x] attic/systems/robotInterfaces/QPLocomotionPlan.h
- [x] attic/systems/robotInterfaces/atlasUtil.cpp
- [x] attic/systems/robotInterfaces/atlasUtil.h
- [x] attic/systems/robotInterfaces/convex_hull.cc
- [x] attic/systems/robotInterfaces/convex_hull.h
- [x] attic/systems/robotInterfaces/test/convex_hull_test.cc
- [x] attic/systems/robotInterfaces/verify_subtype_sizes.cc
- [x] attic/systems/robotInterfaces/verify_subtype_sizes.h

jwnimmer-tri, 2019-10-30:
@RussTedrake @avalenzu @hongkai-dai @siyuanfeng-tri @EricCousineau-TRI any thoughts (or others to CC)?

avalenzu-tri, 2019-10-30:
`world_sim_tree_builder{cc.h}` may be more popular than the rest of these items, as it was the preferred way of putting together RBTs when we stopped using them. Otherwise :+1:

jwnimmer-tri, 2019-10-31:
Yeah, I guess I'll defer `world_sim_tree_builder` to a future pass.  It was going to take some effort to deprecate anyway, since our `examples/kuka_iiwa_arm` still uses it.

jwnimmer-tri, 2019-10-31:
I've opened all of the easy PRs now.  I backed off deprecation for some modules (and unticked their checkboxes above), where there were still uses in-tree or etc.

closed 2019-10-31 by jwnimmer-tri