RobotLocomotion/drake #12289: ci: Need to investigate CDash console debug dump due to errors

EricCousineau-TRI, 2019-10-31:
From Slack:
https://drakedevelopers.slack.com/archives/C270MN28G/p1572401376036300

> jwnimmer-tri 10:09 PM
> The CDash test dump on https://drake-jenkins.csail.mit.edu/job/mac-catalina-clang-bazel-experimental-release/7/console output is very large (20 MB).
> jamiesnape 8:50 AM
> We will look into it. Dumping the whole file on failure definitely is not great.

Perhaps some weird build of cURL on Mac.

EricCousineau-TRI, 2019-11-07:
Zac tried reproducing this locally with cURL, but unable to see the same output. Should try with Catalina-provided cURL?

BetsyMcPhail, 2020-03-11:
The Catalina AMIs use brew installed CMake 3.16.4. Which uses [system curl](https://github.com/Homebrew/homebrew-core/blob/12a076eaf6f0f397283e43a5221d981cf21654c4/Formula/cmake.rb#L36). On Catalina, the system curl is [7.64.1](https://opensource.apple.com/source/curl/curl-118/).

To control the curl output, CMake sets  [CURLOPT_VERBOSE](https://gitlab.kitware.com/cmake/cmake/blob/master/Source/CTest/cmCTestCurl.cxx#L123) to 1 and then  [CURLOPT_DEBUGFUNCTION](https://gitlab.kitware.com/cmake/cmake/blob/master/Source/CTest/cmCTestCurl.cxx#L142) and [CURLOPT_DEBUGDATA](https://gitlab.kitware.com/cmake/cmake/blob/master/Source/CTest/cmCTestCurl.cxx#L157).  The curl output should only displayed if CTest is run with the `--debug` option.

On all Catalina builds, we can see some unexpected curl output: `* Closing connection 0`

This issue was reported on the curl mailing list [here](https://curl.haxx.se/mail/lib-2019-05/0021.html).

The commit that added the extra verbose output was reverted [here](https://github.com/curl/curl/pull/3856) and is in curl 7.65.0

jwnimmer-tri, 2020-03-28:
Three more occurrences:
- https://drake-jenkins.csail.mit.edu/view/Production/job/mac-catalina-clang-bazel-nightly-debug/163/console
- https://drake-jenkins.csail.mit.edu/view/Production/job/mac-catalina-clang-bazel-nightly-everything-debug/164/console
- https://drake-jenkins.csail.mit.edu/view/Production/job/mac-catalina-clang-bazel-nightly-release/165/console

jwnimmer-tri, 2020-03-28:
Homebrew curl https://formulae.brew.sh/formula/curl reports 7.69 as current.  Not sure why this is still happening.

jamiesnape, 2020-03-30:
Because Homebrew CMake uses the system curl. It is unlikely to use its own curl formula for both policy reasons and technical ones relating to circular build dependencies.

jwnimmer-tri, 2020-04-28:
(My mistake was conflating "system curl" with "homebrew curl", leading to my confusion above.  The homebrew version of curl is irrelevant.)

jwnimmer-tri, 2021-11-10:
Closing as "CMake is working as intended".  If Kitware doesn't see fit to apply a bugfix to their Homebrew CMake packages, then who am I to argue?

closed 2021-11-10 by jwnimmer-tri