RobotLocomotion/drake #12303: py solvers: Getting different results when solving in Python 2 vs Python 3?

EricCousineau-TRI, 2019-11-01:
Formulating issue from Slack convo with @pvarin:
https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1572634733031500

> Is there any reason that I should expect the results of Solve() to give me different results in python 2 vs 3 if I can verify that the costs and constraints of the underlying MathematicalProgram are identical in both cases?

EricCousineau-TRI, 2019-11-01:
@pvarin Per the Slack convo, could you post the reproduction code here, if possible?
If you find it easier, you might be able to download an older version of the Drake binaries with Python 2 + 3 packaged on Bionic, e.g. the ones listed for v0.11.0?
https://github.com/RobotLocomotion/drake/releases/tag/v0.11.0
https://github.com/RobotLocomotion/drake/releases/download/v0.11.0/drake-20191002-bionic.tar.gz

RussTedrake, 2019-11-03:
fwiw -- we've noticed that things as simple as the order of the constraints passed to the solver can effect the solver behavior.  (especially, but not exclusively, for the nonlinear solvers)

EricCousineau-TRI, 2019-11-07:
@pvarin Is this still blocking you? Any chance you've had a chance to fix / work around it?

pvarin, 2019-11-08:
Yes, I haven’t had a chance to diagnose yet. I’ve been at IROS all week,
but I’ll post an example next week.

pvarin, 2019-11-11:
Russ guessed it. I was adding some of the constraints to the problem by looping over a dictionary's `items()` method, which resulted in a different constraint ordering when I switched to python 3. So it's not a drake problem after all.

EricCousineau-TRI, 2019-11-11:
Good to know!
I'll close this for now, but please re-open it if you do find different output behavior.

closed 2019-11-11 by EricCousineau-TRI