RobotLocomotion/drake #12306: Not clear how to install drake in installation-from-source instructions

edrumwri, 2019-11-02:
This page (https://drake.mit.edu/from_source.html) tells how to build drake (binaries) but does not tell how to install libraries, binaries, headers, etc. to, e.g., /usr/lib, /usr/bin, etc. Even producing the directories in the same manner as the binary distribution would be okay (not as simple as the typical `make install`, but still fine), but that does not appear to be covered by the documentation for "installation from source".

(Assigning to @jwnimmer-tri, please feel free to reassign)

jwnimmer-tri, 2019-11-04:
I can see that the docs could be improved, I'll see what I can do.

Could you share your actual use case, so that I can make sure it's addressed, and I'm not firing in the dark on what the target is?

However, the short answer is that it's just like any other CMake project -- you do `cmake && make install`, probably from a separate build folder like `cmake ../drake && make install` if you'd like to keep build output separate from the sources.

And of course https://github.com/RobotLocomotion/drake-external-examples shows how to use Drake from another CMake project as an external, though it sounds like you have a different use case in mind.

\CC @jamiesnape FYI

edrumwri, 2019-11-04:
I eventually gathered that the installation process used cmake (from which I also gathered that the `cmake && make install` process was the way to do this) but only after sifting through lots of `./bazel-*` outputs. 

I wasn't aware of https://github.com/RobotLocomotion/drake-external-examples; the `drake_cmake_installed` case describes exactly my use case.

@jwnimmer-tri I think essentially copying the documentation on https://github.com/RobotLocomotion/drake-external-examples (specifically that under "Examples of how to use Drake in your own project") to Drake's installation-from-source would have answered my question.

jwnimmer-tri, 2019-11-04:
Great.  So at a minimum, Drake's docs should hyperlink to `drake-external-examples`.

jwnimmer-tri, 2019-11-04:
(Well, it's linked from the https://drake.mit.edu/ homepage, but the "from source" and "from binary" pages should link to it also.)

jwnimmer-tri, 2019-11-18:
Kicking over to @jamiesnape so as to roll up into the overall installation instruction tune-up of code, docs, etc.

jwnimmer-tri, 2025-01-25:
Fixed in #22461.

closed 2025-01-25 by jwnimmer-tri