RobotLocomotion/drake #12315: Export hydroelastic data to VTK files for ParaView.

DamrongGuoy, 2019-11-05:
We have an in-flight PR #12311 to export basic data for hydroelastic contact model, namely the meshes and the pressure fields.  After this first step, we'd like to extend to slip velocity, normal, etc. See multibody::HydroelasticContactInfo.

Useful technical note from @amcastro-tri :
```
It is very doable to write vector fields with Paraview, it's a matter of using the VECTORS keyword (p. 6 of the file formats reference). With this design you'd need to have a method for each combination of fields you'd ever want to write. For instance, imagine I'd like to debug the pressure field, slip velocity vector field and normals. Then I'd need a new method WriteScalarFieldAndTwoVectorFields()?

That is why I usually split these methods into stages since writing other fields is just a matter of concatenating more to the same file. Typically I write something like this:

write_header(file, title)
write_unstructured_grid(file, mesh)
write_scalar_field(file, field) (sometimes the options for cell center of vertex values is very useful, see CELL_DATA/POINT_DATA. These work for both scalars and vectors and you just place them right before the fields.)
write_vector_field(file, field).
Then for the above example I'd make the following sequence of calls:

Open file
write_header
write_scalar_field(file, pressure)
write_vector_field(file, slip_velocity, kVertexValued)
write_vector_field(file, normals, kCellCenterValued)
Close file.
Maybe a more C++ version of this would be a VtkWriter class with these methods?
```

jwnimmer-tri, 2024-03-16:
I am not sure exactly what the use case is here.  Is the idea that we want to be able to log hydroelastic-related data to files, to help with debugging when things go wrong and/or understanding what's happening?

DamrongGuoy, 2024-03-19:
I was for debugging. I don't need it anymore. I'll close it.

closed 2024-03-19 by DamrongGuoy