RobotLocomotion/drake #12325: Use vtkCapsuleSource directly from VTK

tehbelinda, 2019-11-06:
Use vtkCapsuleSource once it's been pulled upstream into VTK itself, see https://gitlab.kitware.com/vtk/vtk/issues/17720
Then we can undo the third party changes in #12305 

Relates to #10153

sherm1, 2019-11-06:
Assigning to @jamiesnape for now.

jamiesnape, 2019-11-07:
Despite low priority, probably need to do this sooner rather than later to make it into VTK 9 (it is not a difficult or time consuming task, however).

EricCousineau-TRI, 2019-11-21:
Jamie's aiming to submit PR by EOD tomorrow.

EricCousineau-TRI, 2019-12-05:
VTK PR ready; Jamie to add link here.

jamiesnape, 2019-12-11:
https://gitlab.kitware.com/vtk/vtk/merge_requests/6290

jamiesnape, 2020-02-13:
Merged into VTK `master`: https://gitlab.kitware.com/vtk/vtk/commit/e464aa29a09c68aa17891861e309a4df2a7969cb

SeanCurtis-TRI, 2020-02-14:
What's the plan/time frame for the binaries Drake depends on to get updated?

jamiesnape, 2020-02-17:
It's high on my TODO list, so this week.

EricCousineau-TRI, 2020-03-19:
Just waiting to update Drake. May wait for VTK 9.

jamiesnape, 2020-04-03:
FYI VTK 9.0 RC1 is out.

PR #16419 (2022-01-24): Use upstream vtkCapsuleSource
Remove our locally-vendored version of `vtkCapsuleSource` and instead use the upstream version, which is newly available with the switch to VTK 9.

Fixes #12325.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16419)
closed 2022-01-24 by ggould-tri