RobotLocomotion/drake #12330: Initial install issues.

rexbotics, 2019-11-07:
Hi there, 

I have looked around for help in multiple areas and I cant seem to find any answers. I have recently installed Drake on a 18.04.3 machine and I can seem to run the files.

I get 2 errors. After initial installation in the same terminal I try to run the double_pendulum simulation  and i get this 

python double_pendulum/simulate.py
Traceback (most recent call last):
  File "double_pendulum/simulate.py", line 1, in <module>
    from pydrake.all import (AddMultibodyPlantSceneGraph,
  File "/opt/drake/lib/python3.6/site-packages/pydrake/__init__.py", line 29, in <module>
    from . import common
  File "/opt/drake/lib/python3.6/site-packages/pydrake/common/__init__.py", line 1, in <module>
    from ._module_py import *
ImportError: dynamic module does not define init function (init_module_py)

Then if i try to run this is a new terminal, i get this

python double_pendulum/simulate.py
Traceback (most recent call last):
  File "double_pendulum/simulate.py", line 1, in <module>
    from pydrake.all import (AddMultibodyPlantSceneGraph,
ImportError: No module named pydrake.all

I am running from the underactuated/src directory and I have followed all the instructions [here](http://underactuated.mit.edu/underactuated.html?chapter=drake)

Any thoughts? 1 thing I noticed is that when i run:

sudo underactuated/scripts/setup/ubuntu/18.04/install_prereqs

Nothing happens. I am assuming that it doesn't appear try to install anything that I don't already have... but it appears that it is doing nothing at all... Thats the only "error" I can see when going through the install.

Any help would be appreciated! I'll keep scouring the other issues for the time being.

Thanks!

jamiesnape, 2019-11-07:
What version of Python are you using? What git revision or version of Drake?

jwnimmer-tri, 2019-11-07:
The underactuated document says `python double_pendulum/simulate.py`.  That is wrong, it should be `python3 double_pendulum/simulate.py` for starters.  (It might still be broken, but running with python 2 will not work.)

rexbotics, 2019-11-07:
Eureka! So, I tried running python3 in the new window but i never tried it in the initial install window. 

Using python3 in the initial install terminal made it work! Thanks!

As a follow up question though, Why would it not work in a new terminal?

jwnimmer-tri, 2019-11-07:
The command `export PYTHONPATH=/opt/drake/lib/python3.6/site-packages:${PYTHONPATH}` from the instructions only has an effect within one terminal.  It will not persist to new terminals.

rexbotics, 2019-11-07:
Ahhhhh that makes sense. Could I add it to the bashrc file or would i need to run that export everytime?

jamiesnape, 2019-11-07:
Yes, you can add it `.bashrc` if you prefer.

jamiesnape, 2019-11-07:
FYI @RussTedrake Looks like this needs updating? https://github.com/RussTedrake/htmlbook/blob/9aae3451f38599b69eee5b8784bb5e19978c8366/book.js#L220-L233

jwnimmer-tri, 2019-11-07:
Since this report is mostly about install hurdles with the `underactuated.mit.edu` manual, I'll assign to @RussTedrake to decide if there are any documentation improvements to undertake before closing this out.

RussTedrake, 2019-11-08:
@jamiesnape -- i've updated htmlbook to recommend calling `python3` explicitly.

RussTedrake, 2019-11-08:
we have also updated instructions in https://github.com/RussTedrake/underactuated/pull/234.  I believe this is resolved.

closed 2019-11-08 by RussTedrake