RobotLocomotion/drake #12335: Add transmission ratio to revolute joint

sherm1, 2019-11-09:
It is common in actuated joints for there to be a transmission ratio between the input and output shafts (@rcory needs this now). In cases where the transmission details don't matter (typical), we can easily incorporate this ratio into a revolute joint at zero cost. Proposal with three separate tasks (just the first one would be a good start):
- [ ] add `transmission_ratio` (signed) to revolute joints, with a suitable API for setting it programmatically, and unit tests to prove that it actually works!
- [ ] read this from .sdf, where the closest thing is a `gearbox` joint with a `gearbox_ratio` parameter where `output = -gearbox_ratio * input` (see [here](http://sdformat.org/spec?ver=1.6&elem=joint)).
- [ ] read this from .urdf. Looks like the `<mechanicalReduction>` tag in the `<transmission>` element is the closest match (see [here](http://wiki.ros.org/urdf/XML/Transmission)). Not clear to me how the sign is treated.

#### Implementation
Currently the hinge matrix for a revolute joint is `H=[0 0 1 0 0 0]ᵀ`. A reduction of 100:1 would require only that we change to `H=[0 0 0.01 0 0 0]ᵀ`, with a matching change to to the pose calculation (a negative ratio would incorporate a direction change also). This will magically produce the expected reflected inertia, with no bogus messing with mass properties (as suggested, for example, in Featherstone 2008 at the end of §9.6).

This method appears in Jain §12.1.2, page 234.

As an aside, we can do a screw joint with pitch p with `H=[0 0 1 0 0 p]`. That's in Featherstone 2008's Table 4.1, page 79.

sherm1, 2020-02-28:
After some experimentation and discussion with @rcory and @hongkai-dai, here are some details about the implementation of a Drake gear-reducing revolute joint.

One question was whether this should be a separate kind of joint. We propose instead just to add a reduction factor r to the revolute joint constructor. We could do the same for prismatic joints. Note that the reduction factor can be signed, and that values |r| < 1 cause the output shaft to rotate faster than the input shaft. The only restriction is r ≠ 0.

We also have some choices about how to define the generalized coordinate q and generalized velocity v when the reduction ratio isn't 1. For robotics users, it is most natural to think of the input side as a motor (fast, low torque), So:
- the generalized torque τ should be the motor torque
- the generalized velocity v must therefore be the angular velocity of the _motor_
- the generalized coordinate q should be the angle of the output shaft
- therefore the rate q̇ is the angular velocity of the _output shaft_.

Given those choices, we can implement the reduced revolute mobilizer like this: 
```
X_FM(q) = same as plain revolute
H_FM=[ a_F / r,  0, 0, 0]ᵀ    (note Ḣ=0)
N = 1 / r
```
where a_F is the joint axis vector (expressed in the inboard frame F).

I tested this by hacking the existing revolute joint and it appears to work as expected.

sherm1, 2020-02-28:
In a revolute joint, the input and output shafts are aligned. Per discussion with @mitiguy, it is also possible to define a one-dof, constraint-free geared transmission in which the input and output shafts are not aligned. In this case the output axis a_M won't be fixed in the F frame, so we have H=H(q) and can write:
```
X_FM(q) = a bit trickier
H_FM(q) = [ R_FM(q) ⋅ a_M / r, 0, 0, 0]ᵀ    (note Ḣ≠0)
Ḣ_FM = DtF H_FM  (must take time derivative in F frame)
N = 1 / r
```
Not clear if we have an application for this at the moment, but it's a very cool use of the mobilizer equations to shrink a gearbox down to a single ODE!

sherm1, 2020-07-06:
Note (thanks to @yangwill): MuJoCo handles this in a crude way with a joint attribute called ["armature"](http://www.mujoco.org/book/XMLreference.html#joint) which simply adds a fixed value to the mass matrix diagonal for that joint dof's generalized coordinate.

amcastro-tri, 2020-07-09:
As commented in #13657, I believe we'd like:
- JointActuator::motor_inertia and,
- JointActuator::gear_ratio

Those two values is what we'd need to compute proper reflected inertias and place them at the right dofs.

sherm1, 2020-07-09:
@amcastro-tri can you clarify the semantics you have in mind? My understanding is that a motor on a revolute joint has two visible rigid bodies: a _case_ attached to the parent body and an _output shaft_ attached to the child (with internal gearing).  The physical mass properties of those rigid bodies are normally included in the parent and child mass properties. Then the gear ratio incorporated in the H matrix accounts for the reflected inertia. So it's not clear to me what the "motor inertia" would be doing here?

In contrast the proposal in #13657 would be to add user-supplied inertias directly to the mass matrix diagonals, similar to MuJoCo's "armature" feature.

amcastro-tri, 2020-07-09:
> The physical mass properties of those rigid bodies are normally included in the parent and child mass properties. 

Per f2f. What's missing in that picture is a third body, the motor's rotor. That body spins way faster than the output shaft and it effectively has the effect of an added or reflected inertia attached to the output shaft (the child body). In this approximation the small rotor inertia Ir and gear ratio c are swapped by a simple model where there is no motor nor gears but we attach a fat heavy disk having inertia c^2*Im rigidly attached to the outer shaft and spinning with it.

amcastro-tri, 2020-07-10:
ah, great! thanks for double checking @sherm1 and for following up on asking our local roboticists.

yangwill, 2020-07-10:
>Also a quick Slack survey of roboticists here makes it clear that they expect torque control and generalized velocity to represent the _output_ torque and rate, not the torque and rate at the motor's rotor as I modeled above. So I think we should use the c²I approximation which can be added to the mass matrix diagonals or included directly during O(n) recursive sweeps for forward or inverse dynamics. 

This sounds like exactly what I'm looking for, so feel free to close #13657. Before I was concerned that the torque and generalized velocities representing the motor torque and velocity would cause future headaches.

mposa, 2020-07-12:
I'll add my thumbs up to the proposed solution (and choice of coordinates).

mitiguy, 2020-07-14:
I was unclear about this issue.  My understanding (per f2f with @sherm1) is summarized below.  
Feel free to comment is something is amiss (pinging @rcory).
1.  The motor described here consists of a stator and rotor.
2.  The motor's rotor is the "input shaft" to a transmission (e.g., set of gears/belts).
3.  The "output shaft" from the transmission spins relatively slowly relatively to the "input shaft" (motor's rotor).
4.  It would be helpful to have something like a revolute joint that can accounts for some of the inertia effects of the rapidly spinning rotor and gear ratio through the transmission.

amcastro-tri, 2020-07-15:
@mitiguy, for "motor and gearbox combo" as you put it. The discussion above is above a very commonly used model by roboticists, see reference to Featherstone 2008 (chapter 9.6) in #13657. The book describes the approximation in detail.

The biggest thing, summarizing the discussions, is that given that gear ratios are usually very high in robotics, the "effective" inertia of the motor, which is spinning real fast due to this reduction, can now not be ignored. That is, it is pretty much as if I had just placed a heavy inertia wheel attached to the output shaft. Because of this, sims become a lot more stable (typical example; my very low mass finger grippers actuated by a very stiff PID controller. This **today** limits the step size in our sims. This would solve that).

mposa, 2020-09-04:
Any updates/timeline on this issue? @yminchen and @yangwill are finding notable discrepancies between Cassie and sim, with more realistic performance out of MuJoCo which does capture this motor inertia.

We'd look at it ourselves, but I'm concerned that it's pretty far into the MBP weeds where we're not that familiar.

sherm1, 2020-09-04:
Its not on the schedule yet, @mposa. @joemasterjohn what do you think about tackling this smaller project before the joint locking stuff? @mitiguy could advise on unit tests.

edrumwri, 2020-10-30:
I see the reflected inertia being implemented (great!)  Our robot has significant motor inertias and low gear ratios (10:1) but we still want to be able to model both more accurately than is capable at present in sim. So you have another eager customer!

Are you yet able to provide a timeline @joemasterjohn?

mitiguy, 2020-10-30:
Hi Evan. @edrumwri.  Joe is doing great -- and he is in front of this (I am lagging).I am currently on the critical path -- as a reviewer. 
 Thank you for voting in as an important issue.  I will get to it ASAP and make it a top priority.

joemasterjohn, 2020-11-11:
@yangwill @mposa @edrumwri Just wanted to let you know that #14167 merged this morning, and with that the reflected inertia feature is added. Please take it out for a test drive whenever you get the chance, the documentation can all be found in `JointActuator` class. I also have a PR in flight #14323 to add parsing for the added parameters to the SDF and URDF parsers.

yangwill, 2020-11-11:
@joemasterjohn , thank you for implementing this! We've been using this PR in our Cassie simulation and the motions seems qualitatively much more similar to what we're seeing in hardware.

akhilsathuluri, 2021-12-29:
Cannot access `rotor_inertia` and `gear_ratio` through the `JointActuator` class in python. Looks like they are not yet available in python and hence no associated documentation too. Will these be available in python soon or am I missing something?

sherm1, 2021-12-29:
@joemasterjohn want to add these?

sherm1, 2022-11-16:
Only the pybindings are left; replaced this with narrower #18335.

closed 2022-11-16 by sherm1