RobotLocomotion/drake #12336: New issue about running undergraduate class script

ddliugit, 2019-11-09:
Hi, there

I tried to run examples in underactuated class:

`python3 double_pendulum/simulate.py`

Error appears: No module named 'pydrake.all'

Can you help me out?

Another question: will pydrake continue to support python2.x? I really need 2.x package for integrating with other software in the project.

Right now I cannot find python2.x lib after installation.

ddliugit, 2019-11-09:
I exactly follow the installation procedure:

http://underactuated.csail.mit.edu/underactuated.html?chapter=drake

My os version is Mac OS 10.14.6

mpetersen94, 2019-11-11:
Drake no longer supports Python 2 as it is approaching end of life.  To use Python 2 you would have to use the [last tag with Python 2 support](https://github.com/RobotLocomotion/drake/releases/tag/last_sha_with_python_2_support).  As for why python cannot find `pydrake.all`  did you perform the line `export PYTHONPATH=/opt/drake/lib/python3.6/site-packages:${PYTHONPATH}` in the same terminal that you called `python3 double_pendulum/simulate.py` or put that line it in your bashrc?

mpetersen94, 2019-11-14:
@ddliugit Were you able to resolve this issue?  If so, please mark it as closed.  If not, what is still failing?

ddliugit, 2019-11-14:
Sorry, I will test and let you know ASAP.

ddliugit, 2019-11-14:
I tested it with python3 and it works pretty well.

For the matching problem, I solve it by transferring other package from 2 to 3.

Thanks for your help.

closed 2019-11-14 by ddliugit