RobotLocomotion/drake #12344: ComDynamicsFullKinematicsPlanner class

ddliugit, 2019-11-12:
I am working trajectory optimization. Is there any possibility that you will add ComDynamicsFullKinematicsPlanner class in the repo?

hongkai-dai, 2019-11-12:
Thanks @ddliugit for your interest. I am afraid TRI won't add `ComDynamicsFullKinematicsPlanner` in C++ soon, as we don't work on humanoid robot now. Does the Matlab version still work for you?

ddliugit, 2019-11-12:
I download the version"last-sha-with-original-matlab", but I cannot find the instruction of how to install that.

hongkai-dai, 2019-11-12:
The current drake doesn't keep instruction about installing old version.

But in your downloaded version, if you look at the file `drake-last_sha_with_original_matlab/drake/doc/from_cource.rst`, it should have a section `Build the collection", which contains the instruction to build Drake for the version you downloaded. You could check other other revisions of Drake, and they all put the installation instruction in that file.

If you use Ubuntu machine, then I would recommend you follow the instruction on `Build with Make`.

ddliugit, 2019-11-12:
Thank you very much @hongkai-dai. I will go through that carefully.
Actually, I found this repo through your paper titled with "Whole-body Motion Planning with Centroidal Dynamics and Full Kinematics".

Did you use the function in drake to work on atlas' trajectory optimization with Matlab?

hongkai-dai, 2019-11-12:
Thanks for your interest in our work! Yes, the Atlas results from that paper are obtained through the class CoMDynamicsFullKinematicsPlanner in drake with Matlab.

I am not sure if the code still runs. But the class `ComDynamicsFullKinematicsPlanner` can be used as a reference if you want to implement the function yourself.

ddliugit, 2019-11-12:
@hongkai-dai Thanks for your reply. I found a similar little dog [example](https://github.com/RobotLocomotion/LittleDog/tree/b64b2b432756dfd1dc1270700315c9d1977a5818) which is useful for implementation, but I am working on a humanoid robot.
Is there an atlas' trajectory optimization example open with Matlab?
If TRI don't have plan to add trajectory optimization with c++ implementation soon, I can help to contribute that implementation if I have the old code.

hongkai-dai, 2019-11-12:
There is an example in `examples/Atlas/test/testRunningPlanner` which calls CoMDynamicsFullKinematicsPlanner to plan a running gait.

ddliugit, 2019-11-13:
Thank you very much!
I installed the software from [here](https://github.com/RobotLocomotion/drake/releases?after=v0.9.10).
Run the testRunning Planner for about 40 minutes and it is still running now...
My system is Mac Mojave 10.14. Matlab version is 2018b. I7 4.0 GHz
How long usually do you get an answer for the planner?

Invalid MEX-file '/Users/dongdong/Downloads/drake-0.9.9-44f46e2f9dcd1860700b3a65681c09f0ceda1c39-mac/drake/thirdParty/path/lcppath.mexmaci64': dlopen(/Users/dongdong/Downloads/drake-0.9.9-44f46e2f9dcd1860700b3a65681c09f0ceda1c39-mac/drake/thirdParty/path/lcppath.mexmaci64, 6): Library not loaded: /usr/local/lib/x86_64/libgfortran.3.dylib
  Referenced from: /Users/dongdong/Downloads/drake-0.9.9-44f46e2f9dcd1860700b3a65681c09f0ceda1c39-mac/drake/thirdParty/path/lcppath.mexmaci64
  Reason: image not found.

Another reason I think may be caused by above error.

hongkai-dai, 2019-11-13:
I never used precompiled version of Drake.

I use Drake on Ubuntu with SNOPT. It is possible that the pre-compiled drake in MATLAB doesn't use SNOPT. It should took just a minute to finish the trajectory optimization.

The error you got `Cannot find required pod snopt` means that Drake uses another nonlinear solver, probably fmincon, which is very slow for this trajectory optimization problem.

The error 
```
Invalid MEX-file '/Users/dongdong/Downloads/drake-0.9.9-44f46e2f9dcd1860700b3a65681c09f0ceda1c39-mac/drake/thirdParty/path/lcppath.mexmaci64': dlopen(/Users/dongdong/Downloads/drake-0.9.9-44f46e2f9dcd1860700b3a65681c09f0ceda1c39-mac/drake/thirdParty/path/lcppath.mexmaci64, 6): Library not loaded: /usr/local/lib/x86_64/libgfortran.3.dylib
  Referenced from: /Users/dongdong/Downloads/drake-0.9.9-44f46e2f9dcd1860700b3a65681c09f0ceda1c39-mac/drake/thirdParty/path/lcppath.mexmaci64
  Reason: image not found.
```
means that it cannot find the `pathlcp` solver. This solver is used in simulating rigid body dynamics. It is not used in trajectory optimization.

Sorry I can't offer much help to fix the SNOPT issue. I would recommend to use the implementation `CoMDynamicsFullKinematicsPlanner` as just as a reference. You could try to download snopt from their website (some universities might have an academic versions, and I think SNOPT still provides a free version with limits on 300 decision variables and constraints). You could then add the snopt folder to your MATLAB path (make sure that you have compiled snopt and obtained a mex file for MATLAB).

ddliugit, 2019-11-14:
Thanks for your detailed reply!

I got a copy of SNOPT 7.7 by sending a request to UCSD, but I got the same error reported in #6123 , then I just followed the procedure reported in #1488.
However, I got the same error reported in #6123 again. 
```
Error using snoptmex
Wrong number of input arguments

Error in snopt (line 250)
[x,F,inform,xmul,Fmul,xstate,Fstate,itn,mjritn] = snoptmex( solveopt, x, ...

Error in NonlinearProgram/snopt (line 1411)
        [x_free,objval,exitflag,xmul,Fmul] = snopt(x0_free, ...

Error in NonlinearProgram/solve (line 980)
            [x,objval,exitflag,infeasible_constraint_name] = snopt(obj,x0);

Error in testRunningPlanner (line 430)
  [x_sol,~,exitflag] = cdfkp.solve(x_seed);
```
In #6123, @GuiyangXin just switched to 7.2 version to solve that problem.

Is there any way to fix the problem of snopt paramete with latest version?

ddliugit, 2019-11-14:
I figured that out and wrote a small patch code in snopt.m. It works!

However, the solution failed!

```
Warning:   13 nonlinear infeasibilities minimized
 
> In NonlinearProgram/mapSolverInfo (line 1551)
  In NonlinearProgram/snopt (line 1427)
  In NonlinearProgram/solve (line 980)
  In testRunningPlanner (line 430) 
Elapsed time is 20.314129 seconds.
Error using testRunningPlanner (line 435)
Trajectory optimization failed. exitflag = 13 
```

Do you have any idea why it fails?

hongkai-dai, 2019-11-14:
Thanks for the investigation! Snopt 7.7 and snopt 7.2 have different implementations. We observed that the nonlinear optimization generates different results (solved to optimal vs local infeasibility) after we upgrade our snopt version.

You could try to tweak the initial guess of the optimization problem. Snopt is a nonlinear solver, and the nonlinear solver performance depends heavily on the initial guess.

hongkai-dai, 2019-11-14:
BTW, if you have `iAfun, jAvar, Avals`, then you could set `A` as a struct, and
```
A.row = iAfun;
A.col = jAvar;
A.val = Avals;
```

ddliugit, 2019-11-14:
Thanks. 

I just tested it with struct and it runs, too.

Make up the remaining code for other people in case they need to use the old version of drake with new SNOPT. 

SNOPT 7.7
Drake v0.9.10.
environment: macOS

In the NonlinearProgramming.m: insert following code after line 1409:
```

        A.row = iAfun;
        A.col = jAvar;
        A.val = Avals;
        
        G.row = iGfun_free;
        G.col = jGvar_free;
```

The other way is insert the following code after line 420 in snopt.m in snopt_matlab wrapper.
```

elseif nargin == 17,
  ObjAdd = varargin{1};
  ObjRow = varargin{2};
  valA   = varargin{3};
  iAfun  = varargin{4};
  jAvar  = varargin{5};
  iGfun  = varargin{6};
  jGvar  = varargin{7};
```

I just used your solution as the initial value and it works through(silly method, just want to test if the program runs well)

Tweaking the initial guess seems quite difficult and needs special technique. The default option is inserting linear points between initial value and final value.  Intuitive idea to me is use spline line to generate these points between start and end points. 

Do you have any good idea to tune that?

hongkai-dai, 2019-11-14:
Thanks for the fix!

Right as you said, tweaking the initial guess can be difficult. There is not a principled approach to tweak the initial guess. I mostly do it by trial and error. For this specific problem, you could do an interpolation between the initial state and the final state, and then set the contact forces to be roughly balancing the gravitational forces.

As your code runs, is it OK if we close this issue? We can re-open it if you run into more problems.

ddliugit, 2019-11-15:
I truly appreciate your patient help!
Thanks again! @hongkai-dai

ddliugit, 2019-11-18:
I have another question, can I use ComDynamicsFullKinematicsPlanner in an implicit contact trajectory optimization problem?

I checked the testRunningPlanner, it seems a sequential contact order will be indicated before the optimization problem, namely, the contact is explicit.

hongkai-dai, 2019-11-18:
In that folder `drake/solvers/trajectoryOptimization`, there are several files start with `Complementarity`, for example `ComplementarityStaticContactConstraint.m`. These classes impose the complementarity constraint so that you can do implicit trajectory optimization. In line 331 of ComDynamicsFullKinematicsPlanner.m, it adds the complementarity constraint, if you specify the complementarity_flag = True.

ddliugit, 2019-11-19:
@hongkai-dai Thanks for your reply! 
For complementarity function, I ignore the abstract method in `SimpleDynamicsFullKinematicsPlanner`. I checked through the complementarity and find these complementarity constraints are static, but if I want to use it in sliding case of friction cone, will that work?

hongkai-dai, 2019-11-21:
Sorry for my belated reply. I haven't implemented the sliding friction version yet. You could add a complementarity constraint on the sliding friction. The idea is that the if the sliding velocity is zero, then the force is within the friction cone. If the sliding velocity is non-zero, then the tangential force should be in the opposite direction of the sliding velocity, and the friction force is on the boundary of the friction cone. You could refer to Michael's paper "A Direct Method for Trajectory Optimization of Rigid Bodies Through Contact" for a detailed formulation.

ddliugit, 2019-11-24:
Thanks for your reply. I go through that paper and it's clear now.

I adjust the initial pose of the robot (not your solution) and it can calculate an answer now!

But when I use the controller `runAtlasRunning.m` to verify my solution, I found `bodyMotionControlmex` file is missing.

Currently I can only found the `*.cpp` version instead of its compiled version. I plan to recompile it and the error where I am stuck is like so:

```
Undefined symbols for architecture x86_64:
  "_mexCallMATLABWithObject", referenced from:
      isa(mxArray_tag const*, char const*) in drakeMexUtil.cpp.o
      mexCallMATLABsafe(int, mxArray_tag**, int, mxArray_tag**, char const*) in drakeMexUtil.cpp.o
  "_mexCallMATLABWithTrapWithObject", referenced from:
      mexCallMATLABsafe(int, mxArray_tag**, int, mxArray_tag**, char const*) in drakeMexUtil.cpp.o
  "_mexErrMsgIdAndTxt", referenced from:
      mexCallMATLABsafe(int, mxArray_tag**, int, mxArray_tag**, char const*) in drakeMexUtil.cpp.o
...

ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
Do you know the reason?

ddliugit, 2019-11-25:
I made a mistake that I should use the flag while generating make file.  
`-DCMAKE_BUILD_TYPE=Release`, but other error appears still.

I fix arm, head joints, then remove their constraints(gazing constraint, etc) and only keep the leg part. It doesn't work well. 

I guess you add the gaze constraint because we can limit the movement of the body which can help the snopt solve the problem quickly?

For your last answer, I am not clear about balancing the gravity with contact force while setting up initial solution.

Another issue is that I am not clear about your setup in the initial seed(line 405 in testRunningPlanner.m):

`lambda_seed = sqrt(2)/24;`

Is this code relevant to the above idea?

> Thanks for the fix!
>
> Right as you said, tweaking the initial guess can be difficult. There is not a principled approach to tweak the initial guess. I mostly do it by trial and error. For this [...]

hongkai-dai, 2019-11-26:
Sorry I don't quite understand the linking error, it complains about `_mexCallMATLABWithObject`, but this is a MATLAB built in function.

For `lambda_seed = sqrt(2)/24`, the goal is to have the contact force to balance the gravitational force in the initial guess.  There are 8 contact points on two feet, each contact point has a linearized friction cone of 3 edges, each linearized friction cone has a friction coefficient of 1 (namely the angle of the friction cone is 45 degrees, and cos(45 degree) = 1 / sqrt(2)). So by setting `lambda_seed = sqrt(2) / 24`, the total contact force is 8 * 3 * lambda_seed  * cos(45 degree) * mg = mg, so as to balance the gravitational force.

To make the decision variable values on the same scale, that `lambda` is not the contact force, but the weight of each edge of the friction cone, **normalized by mg**.

> I guess you add the gaze constraint because we can limit the movement of the body which can help the snopt solve the problem quickly

I added the gaze constraint since otherwise there is too much motion on the upper torso, I don't want the upper torso to shake a lot during running.

ddliugit, 2019-11-27:
@hongkai-dai .Thanks for your detailed reply, I see. 

I assume lambda is contact force after reading your paper, but in your code it is the coefficient of the linearized edge, which you use beta_ij to indicate in the paper.

I found errors while compiling it on Mac, but it seems better on Ubuntu(No error appears right now except Gurobi solver is not installed for that library). Which version did you use with Ubuntu then? I will prefer to use the same version to prevent further errors from happening. 

I import a small robot model different from Atlas. I try to find the solution, but it doesn't work. The following is the result I got:

[image]

In the picture, the solver gave me two basins, which means that robot just jumps up, then jumps down, and jumps up again, but while running it should be one basin like the solution given by Atlas. Is there any idea to remove it by tuning the initial solution? Would adjusting the knots of touch toes and heels affect the result?

I tried it many times and it failed. I found the final result of atlas is kind of symmetry, like a parabola. Is there any reason why this works? And how to tune it to become that?

Another confusing part is that I don't know how to define the speed and stride length of the robot.

For the initial pose, I just copy the pose of Atlas to the new robot, would that matter? 

What is the max iterations did you usually set for the solver when you adjust the parameters or initial solution?

Do you refer the log of snopt solver to tune the parameter? 

[image]

Sorry for so many questions. Thanks again!

hongkai-dai, 2019-11-27:
> Which version did you use with Ubuntu then?

I think I was on Ubuntu 12.04 LTS when I wrote the paper. The code should work on other versions of Ubuntu also.

> In the picture, the solver gave me two basins, which means that robot just jumps down, then jumps up, and jumps down , up again, but while running it should be one basin like the solution given by Atlas. Is there any idea to remove it by tuning the initial solution? Would adjusting the knots of touch toes and heels affect the result? Another possible reason is caused by the initial foot height?

The peak between these two basins are quite low, just 1 cm above the basins, does this peak matter? I don't know your robot, but I don't see immediately why you want to remove this peak.

> I tried it many times and it failed. I found the final result of atlas is kind of symmetry, like a parabola. Is there any reason why this works? And how to tune it to become that? 

I don't think we explictly added any constraints to make it symmetric or parabola. The tweaking we did are mostly on the kinematic part of the robot, like the using a gaze constraint on the upper torso to prevent it from swinging.

> Another confusing part is that I don't know how to define the speed and stride length of the robot.

You could define the speed as the final CoM translation divides by the total time, you could then impose a constraint on this ratio (by writing your own constraint). For the stride length, you could impose a WorldPositionConstraint on the foot contact location.

> For the initial pose, I just copy the pose of Atlas to the new robot, would that matter?

That is probably not a good thing, if the two robot have different DoFs. I would solve an inverse kinematics problem to find an initial pose of your robot, that the robot is in static equilibrium.

> What is the max iterations did you usually set for the solver when you adjust the parameters or initial solution?

The setup for the solver options (like iterations) are in line 419 of testRunningPlanner.m.  You could see I didn't care too much about the optimality (the optimality tolerance is 5e-4).

> Do you refer the log of snopt solver to tune the parameter?

Referring to the log is helpful. By looking at the merit function or "Feasible" part, I can see how close I am to the feasibility. If the solver spends many iterations without decreasing the "Feasible" tolerance, I will just use another initial seed as the starting point. Another useful debugging approach is to add a callback to your optimization, that in each iteration it plots out something (like CoM trajectory).

ddliugit, 2019-12-01:
@hongkai-dai Thanks!

Solving inverse kinematics for static equilibrium really gave me a good result. The feasible error is from 9.8e-2 to best 1.9e-3, but it still cannot achieve a desired trajectory.

I also recorded the result after the tolerance achieved 1.9e-3. It seems the foot just move randomly, which is different from Atlas solution. 

Here is the picture:

For own robot example:
[image]
[image]
[image]

For Atlas:
[image]
[image]

My robot foot doesn't align towards forward like Atlas. 

This robot doesn't have arms and I reduced the half periodic constraint with arms. Would that cause the random movements of feet.

Another problem is swing moving of the torso. Because there is no arm, the gaze constraints cannot be applied. I just set bounding box constraint for rotation variable of the torso, would that be good?

I just tried that method, it affects the performance of the solver, which stuck at a certain solution early.

Do you have any good idea to solve these problems?

ddliugit, 2019-12-02:
For the rotation of foot while touching ground, I guess it is caused by missing of `equal matrix for the torso` in `half_periodic_constraint = LinearConstraint(lb,ub,[symmetric_matrix;equal_matrix]);`I deleted that before and the rotation of feet seems caused by the rotation of body.

After adding gazing constraint and making up half periodic constraint, the problem of moving feet randomly on the ground is better.

Another issue is that the solver always tries to find a solution with larger height of center of mass than initial seed, which makes the robot smooth the initial pose totally, i.e. no knee bending, both of  legs straight during running.

[image]

The actual robot's com is about 0.31 before running the optimization. However, the optimizer always stays about 0.365 even at the beginning, which will make the robot do inverse kinematics to totally change the initial seed. 

The desired solution is that optimizer should find solution under initial com height while robot touches the ground.

I may miss something or happen to add some constraints which I don't notice. What do you think?

Below is initial pose and snap shots for one solution which is not complete. 

[image]

[image]

hongkai-dai, 2019-12-02:
> which makes me be curious why Atlas' feet always fixed on the ground while touching and always in the one direction

There should be a kinematic constraint that the feet doesn't move on the ground. Drake's Atlas example is solved successfully, so the foot don't move. In your case, as the problem is not solved to a feasible solution yet, the output trajectory violates this "foot no move" constraint.

You could enforce the constraint "feet always in one direction", by imposing a `WorldGazeDirConstraint`

> Instead of gazing constraint I just set bounding box constraint for three rotation variables of the torso, namely, q[4:6,N], would that be good?

That might work for your robot. From your plot, it seems that that your torso is your floating base. But q[4:6] is the Euler angles of the torso, which is not exactly the same as the gaze constraint (but might prevent the torso swinging problem). I would suggest to start with a minimal set of constraints (like the foot should touch the ground at the beginning), and only add other constraints (torso no swinging, foot in one direction, ect) when you see bad behaviors from the output trajectory, so as to present these bad behaviors by imposing constraints.

> I am not sure how to setup the translation of torso(base of robot, since atlas has extra DoFs in body), and you seems to setup the torso translation to make CoM be zero at x and y direction in the world coordinate system. Is there any reason for that? 

You could set a bounding box constraint on the x/y position of the torso at the beginning, and a bounding box constraint on the x/y position at the end of the trajectory. I set the constraints on the CoM x/y position, but you could also try to bound the torso position. 

> About the height of torso(z direction), because atlas begins at certain height above the ground, do you remember how to set it up initially?

I think I solved an IK problem to find the initial pose of Atlas. In that IK problem, I have the constraint that
1. The CoM is within the support region of the foot (Check QuasiStaticConstraint).
2. The knees are bent (we didn't know how to control Atlas with straight knees back then).

> Another issue is that the solver always tries to find a solution with larger height of center of mass than initial seed, which makes the robot smooth the initial pose totally, i.e. no knee bending, both of legs straight during running.

You could add a constraint that the knee should bent during running.

jwnimmer-tri, 2022-04-28:
@hongkai-dai Could you help us clarify the topic of this issue?  What are the victory conditions necessary to close it?  Is this a feature request?  If so, what is the feature?  Is there any chance we will work on it?  I'm curious whether we should close this issue.

hongkai-dai, 2022-04-28:
I consider this issue closed. We have [CentroidalMomentumConstraint](https://github.com/RobotLocomotion/drake/blob/master/multibody/optimization/centroidal_momentum_constraint.h) inside Drake, and Russ also has an example of little dog doing centroidal dynamics planning in underactuated repo https://github.com/RussTedrake/underactuated/blob/master/examples/littledog.ipynb

closed 2022-04-28 by hongkai-dai