RobotLocomotion/drake #12359: Failure at multibody/plant/multibody_plant.cc:1640 in CalcImplicitStribeckResults(): condition 'info == ImplicitStribeckSolverResult::kSuccess' failed.

54WC, 2019-11-16:
Sometime, drake running fail due to bad parameter setting, the running is broke and come out a failure message on the bash terminal. like:Failure at multibody/plant/multibody_plant.cc:1640 in CalcImplicitStribeckResults(): condition 'info == ImplicitStribeckSolverResult::kSuccess' failed.

In case we running a parameter search, we often use for-loop to test a lot of conditions. 
Could we just skip the bad condition and test next condition rather than totally break the code running ?

jwnimmer-tri, 2019-11-18:
\CC @amcastro-tri @edrumwri @sherm1

edrumwri, 2019-11-18:
@54WC this means that the Simulator is not able to simulate the MultibodyPlant at this step size. Since this is an assertion, there's no way to catch the error (except trapping the signal, I suppose).

Some simple options are:
1.  Simulate at a smaller step size. There should be a sufficiently small step size at which point this failure won't occur (barring some NaN parameter settings).
2. Simulate using a continuous-mode MultibodyPlant (i.e., construct with a zero step size). The simulation will likely run more slowly, but this particular abort will no longer happen. 

@sherm1 it would be prudent for MBP to return a failure status in its discrete state update method (in place of the abort), which @54WC could then use to try simulating with a smaller step size.

jwnimmer-tri, 2019-11-18:
As a workaround in the meantime, why not change the assertion to an exception?  Then, users (especially, pydrake users) could catch the exception and move on to the next scenario.

54WC, 2019-11-18:
> @54WC this means that the Simulator is not able to simulate the MultibodyPlant at this step size. Since this is an assertion, there's no way to catch the error (except trapping the signal, I [...]

Thanks!  If there is a way to catch the error,  a " very bad reward" can be feedbacked to the action value in reinforce learning model.   Then the program will learn and go for another save way in the next episode.

amcastro-tri, 2019-11-18:
As @edrumwri pointed out, this failure means the time step is too large for the given simulation conditions and causes the solver to diverge. The results are no longer valid and then we abort. I agree we could make this an exception @jwnimmer-tri.

amcastro-tri, 2019-11-18:
See #12361. Also @54WC, I see you are running a version of Drake almost two month old (if not older).

54WC, 2019-11-19:
Yes, I start to run some code on Drake after  finishing Underactuated Robotic Spring 2019 course.  Actually, I followed Underactuated robotic course every year since 2015.  It is amazing but difficult for me. 
I expect I can build my underactuated robot someday at my slow pace.

closed 2019-11-20 by jwnimmer-tri