RobotLocomotion/drake #12376: Transition to starlark versions of native build rules

jamiesnape, 2019-11-20:
Using https://github.com/bazelbuild/bazel-federation may be the best bet. Transitioning to `rules_python` is the immediate priority since the native builds are already deprecated.

jamiesnape, 2019-11-20:
Probably should fix the `rules_python` in tandem with #11380.

jwnimmer-tri, 2020-04-28:
We have rules_python already converted.  We still have to do rules_cc and rules_java.

jwnimmer-tri, 2022-03-16:
The separate starlark rules_cc is far from complete, and is generally a regression from the native ones.  We'll wait to switch until Bazel deprecates the native ones.

For rules_java, we use it so little that there's no real need to switch anything unless Bazel deprecates the built-in ones.

closed 2022-03-16 by jwnimmer-tri