RobotLocomotion/drake #12382: mac: Ensure that we match clang-format with Ubuntu when Homebrew drops llvm@6?

EricCousineau-TRI, 2019-11-21:
Per f2f with @jamiesnape, we have `clang-format-8` on `bionic-updates`, so we could upgrade both.

EricCousineau-TRI, 2020-01-03:
May have gotten bitten by this in this job: https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=1257066
For PR #12505

jamiesnape, 2020-01-03:
No, this issue would be seen an error on a (most likely) unprovisioned build saying that `clang-format` could not be found.

soonho-tri, 2020-01-03:
I wonder if https://github.com/grailbio/bazel-toolchain can be a solution to this problem.

> The LLVM toolchain also provides several tools like clang-format. You can depend on these tools directly in the bin directory of the toolchain. For example, @llvm_toolchain//:bin/clang-format is a valid and visible target.

jwnimmer-tri, 2020-04-28:
For Ubuntu 20.04, I'm going to recommend that we upgrade to clang-format-9, so that we have:
- an identical version on both 18.04 and 20.04, and
- a version that does not depend on python2 (clang-format-6.0 depends on python2)

jwnimmer-tri, 2020-05-14:
We are using clang-format 9 on master now, both on Ubuntu and macOS.  It seems to me like that should suit us for a while.

closed 2020-05-14 by jwnimmer-tri