RobotLocomotion/drake #12398: How to handle the format returned from the optimizer solver, like <Expression "3">?

vinbo, 2019-11-24:
The nonlinear programming solver returned this format:
initial state: [<Expression "-2"> <Expression "0.10000000000000001">
 <Expression "-0.10000000000000001"> <Expression "3">]

although I was expecting float.  Its type is <type 'numpy.ndarray'>, which seems right.
How do I convert this format to float, please? Thanks

hongkai-dai, 2019-11-24:
Could you paste your code here so that we can reproduce your output?

vinbo, 2019-11-25:
> 
> 
> Could you paste your code here so that we can reproduce your output?

Thank you. I pasted it in the main post.

hongkai-dai, 2019-11-25:
I can't run your code, as I only have pydrake compiled in python3. But I think the problem is that `initial_state` is a numpy array of float, it is not a numpy array of symbolic variables or symbolic expressions. So when you call `result.GetSolution(states_over_time)`, it first tries to convert the numpy array of `[-2., 0.1, -0.1, 3]` to a numpy array of symbolic expressions (with constant expressions).

The solution would be to not to call `result.GetSolution(states_over_time)`, but `result.GetSolution(states_over_time[4:])`. You already know the value of initial state as `x0`, so you don't need to get its value from `result`.

vinbo, 2019-11-25:
> I can't run your code, as I only have pydrake compiled in python3. But I think the problem is that `initial_state` is a numpy array of float, it is not a numpy array of symbolic variables or [...]

Thanks. It works. But not result.GetSolution(states_over_time[4:]) , but result.GetSolution(states_over_time[1:]).
And I have a serious suggestion: Why not write the document more concrete and friendly? It is really hard to read. Take this post as example, even you were making mistakes about wether the index should be [4:] or [1:], because the document does not say anything about it. More examples are needed in the documents.

hongkai-dai, 2019-11-25:
> Why not write the document more concrete and friendly

Could you point out which function needs better documentation?

We do have a very basic tutorial on Mathematical Program https://nbviewer.jupyter.org/github/RobotLocomotion/drake/blob/master/tutorials/mathematical_program.ipynb. We will add more tutorials in the future.

vinbo, 2019-11-26:
> Could you point out which function needs better documentation?

The tutorial link is good, but the document https://drake.mit.edu/pydrake/pydrake.solvers.mathematicalprogram.html, or all the documents down here are hard to read

vinbo, 2019-11-26:
>
> Could you point out which function needs better documentation?
> 
here is another example of the document being too concise to read
https://github.com/RobotLocomotion/drake/issues/12410

jwnimmer-tri, 2019-12-17:
It looks like this has reached a conclusion.  @hongkai-dai If there is something still to do here, please re-open and clarify what is the acute bug report.

closed 2019-12-17 by jwnimmer-tri

vinbo, 2019-12-21:
another trick to handle the problem is here:
https://stackoverflow.com/questions/59335527/runtimeerror-you-should-not-call-bool-nonzero-on-formula/59433250#59433250