RobotLocomotion/drake #12399: jupyter: Need to make it easy to review notebooks in Reviewable, commit to Git, etc.

EricCousineau-TRI, 2019-11-25:
Hoisting this from Anzu issues; this solution states that we should try to just save to `*.py` files; however, that interrupts ease-of-use with other tools like Binder, so this may not be sufficient:

> At present, the only save/load format we have for Jupyter notebooks is the default JSON format.
> It would be much more ideal if the format saved to Git is a Python file that's easier to review, as dealing with JSON data as code is a pain.
> 
> I had originally tried to mess around with Git hooks and stuff to do this type of stuff, but it's a can of worms.
> 
> One solution path is to write a notebook file format s.t. it will save/load directly to this format. This would be much better on many fronts (workflow, ease of use, etc.).
> 
> Another solution is to try and inspect the save/load hooks for Jupyter notebook, and see if there are ways to do file conversion. @thduynguyen mentioned that there is a VSCode plugin, so we can look at it and see if there's a way to incorporate that:
> https://code.visualstudio.com/docs/python/jupyter-support
> https://github.com/Microsoft/vscode-python/tree/master/src/client/datascience/jupyter - note - it's typescript :(
> 
> Some requirements:
> * Round-trip should be idempotent conversion, modulo minor states (like scrolling, edit mode for Markdown, etc.)
> * Markdown cells should automatically
> * Saving the notebook should automatically write to the source file being edited

EricCousineau-TRI, 2019-12-05:
@thduynguyen suggested: https://www.reviewnb.com/

@jamiesnape How feasible might it be to incorporate this?
@jwnimmer-tri Does this seem like it would complicate the review process? (doesn't seem like it has the super-awesome discussion features that Reviewable has)

I'd be up for doing a small pilot session, if someone else has the patience to try it out.

jwnimmer-tri, 2019-12-05:
The first solution I would try is not a new app, but just committing the notebook's python to git, with a linter to remind people to keep it up to date.  (The linter could also check for empty cells, while we're at it.)

EricCousineau-TRI, 2019-12-05:
Ah, makes more sense, will try that out!

EricCousineau-TRI, 2020-07-22:
FWIW I'd rather start off with code that can auto-reformat notebooks when applicable.
An example of using isort + black (relates #13709):
https://github.com/EricCousineau-TRI/repro/blob/3706e9c9e3395d252c682402879b662291bc951d/python/bazel_pystyle_black_isort/python_lint.py#L71-L99

jwnimmer-tri, 2021-11-11:
Close as duplicate of https://github.com/Reviewable/Reviewable/issues/550 and https://changelog.reviewable.io/jupyter-notebook-reviews-question-212809.  Let's just wait for a first-party solution to this, instead of rolling our own.

closed 2021-11-11 by jwnimmer-tri