RobotLocomotion/drake #12400: MultibodyPlant needs unit tests for checking correctness of MapVelocityToQDot and MapQDotToVelocity

edrumwri, 2019-11-25:
During platform review of #12355, it was observed that there is only a test for consistency of mappings between MapVelocityToQDot() and MapQDotToVelocity().

amcastro-tri, 2019-11-25:
@edrumwri, what do you have in mind to test this? any suggestion?

edrumwri, 2019-11-25:
Sure- I think an articulated body with at least two mobilizers would ensure that the mapping happens using the correct ordering. That would still be more of a consistency check: @sherm1 asked for correctness, except that correctness testing happens in the individual mobilizers so I would be fine not doing so in MBP unit testing (but if you wanted to make @sherm1 happy, you could construct a body in such a configuration that it's qdot and v mapping would be simple to determine and verify as well).

amcastro-tri, 2019-11-25:
We need something more concrete, with an analytical known solution, instead of us copy/pasting what the mobilizer implements which I believe would not add any value. 
Let me then paraphrase my previous question; do you have in mind a case with analytical/known solution that we can use to test these mappings?

sherm1, 2019-11-25:
I think a joint with a quaternion that is conveniently posed along an axis would be enough for an easy test for qdot <-> ω interconversion and also would verify that the change in length is handled right. @mitiguy would likely have better ideas and some MG-produced results we could bake in.

amcastro-tri, 2019-11-25:
I like the idea of producing results with another software. Thanks

closed 2024-03-20 by jwnimmer-tri