RobotLocomotion/drake #12401: mbp: For quadrotor-modeling workflows, would be nice to have 2DOF floating base coordinates and "thruster" actuators

EricCousineau-TRI, 2019-11-25:
(My attempt to transcribe a Slack convo with @mposa, regarding [today's Drake Developers meeting](https://docs.google.com/document/d/15iNBw_2O3TGbThHr6wlJNKctHSY5bEvt1rLhl5obZao/edit#heading=h.w1yphvs70spa))

Michael's students are modeling a 2D quadrotor, and have found it to be not necessarily straightforward to model a 2D floating base with quadrotor thruster actuators.

Since we only have 3D floating base mobilizers at present, they could not leverage our current floating base setup. Instead, they added two prismatic joints (x, y) and one revolute joint (z), and (I believe) with made-up links in between those virtual joints.

To model thrusters, they made those virtual joints actuated, and then mapped their model inputs (thrust) to the virtual joint inputs with a mapping done by another system block.
For this workflow, it would be more ideal to have something that could more explicitly and directly model thrusters.

From the meeting, one possible workflow for actuation was to add a `SpatialForce` input, connected via `MultibodyPlant::get_applied_spatial_force_input_port()`, and still do the mapping there. However, this does not address the planar floating base DOF problem.

@sherm1 Can I ask who would possibly be able to work this into their queue? (I won't be, at least for the remainder of this quarter.)

@rcory Can I ask how y'all model the planar floating block for your dexterous manipulation example?

sherm1, 2019-11-25:
Not sure what is meant by a "2D quadrotor"?

In any case we should have a planar joint with x-y-θ generalized coordinates for convenience. I'm not sure this would be a panacea though:
- Drake is a 3D code so gives free bodies 6 dofs -- so the quadrotor input would have to specify the planar joint to World explicitly -- it wouldn't get selected as a default floating base.
- .sdf does not seem to support planar joints at all -- see the [joint types](http://sdformat.org/spec?ver=1.6&elem=joint#joint_type) listed here.

It appears that all the previous .urdf support for quadrotors was through custom Drake tags, see the schema [here](https://github.com/RobotLocomotion/drake/blob/master/doc/drakeURDF.xsd) (look for "propellor" (sic) for example). I'm not sure the sdf specification has anything comparable, although it does seem to support "wind".

I'm not seeing yet that adding a planar joint would go very far to making the quadrotor problem tractable for Mike's students. There was apparently a lot of support added to Drake-Matlab urdf -- I'm guessing we wouldn't want to handle that the same way in modern Drake. Instead, I would expect having a few pre-defined Systems that could be wired in to produce the same effects would be better?

sherm1, 2019-11-25:
Filed issue #12404 to track requests for additional joint types.

rcory, 2019-11-25:
>  they added two prismatic joints (x, y) and one revolute joint (z), and (I believe) with made-up links in between those virtual joints.

@EricCousineau-TRI the manipulation example does exactly this.

RussTedrake, 2020-04-21:
For reference, here is the old matlab implementation of the propellor:
https://github.com/RobotLocomotion/drake/blob/last_sha_with_original_matlab/drake/matlab/systems/plants/RigidBodyPropellor.m

RussTedrake, 2020-04-21:
Since #12404 is tracking the joint type request, I will rename this to focus on the propellor only.

RussTedrake, 2020-04-21:
My proposal is to write `PropellorSystem` which has:
- a constructor takes MBP body index and a `RigidTransform` from body coordinates to the propellor (+ thrust and moment ratios). 
- an input port for propellor thrust command
- an input port for `geometry_pose` (to be wired to the MBP `geometry_pose` output)
- an output port for `ExternallyAppliedSpatialForce` input.
- a `CalcOutput` method which transforms the thrust/moment from actuator coordinates to world coordinates.

Since we typically want to have multiple propellors on a single system, I think I will also end up writing a `ExternallyAppliedSpatialForceMux` that accumulates multiple spatial forces into the one MBP input port.

I know this isn't going to fit nicely into our current parsing model.  But i think it will be immediately useful to some folks even if Propellors can only be added programmatically.  I'm pretty sure that the `ForceElement` workflow does not work for this (because it does not offer an external input).  I'm sure that I could wedge it in, but I think at some point we have to stop cramming things into MBP and leverage things like this at the system's framework level.

@sherm1 , @amcastro-tri ... would love to get your thoughts before I dive in.

sherm1, 2020-04-21:
My first thought is: Please spell it "propeller"! :)

sherm1, 2020-04-21:
I like this proposal and also feel that the System framework ought to provide nice "blocks" that can be wired into MBP to apply forces. A LeafSystem is currently much more powerful than a ForceElement -- it can define input ports as you mentioned but also state and differential equations, parameters, event handlers, etc. We have provision for tallying up the energy and power from separate systems.

We had some discussion about this regarding the DoorHinge which we modeled with a ForceElement -- it remains unclear why. I asked Grant and he said it hadn't occurred to him to do it as a separate system. It would have been a plausible alternative though not _obviously_ better.

Parsing is an issue. If an sdf can specify a propeller we should be able to build a proper Drake Diagram from that specification. But we already parse into an MBP+SG diagram so it doesn't seem inconceivable that the parser could learn about more Drake force model systems and include those too.

amcastro-tri, 2020-04-21:
Love the idea @RussTedrake. +1 for propeller :-)

RussTedrake, 2020-04-21:
Ok, thanks.  I will PR something in the next day or two.  Do you both also agree on the need for a system to accumulate multiple external forces?

jwnimmer-tri, 2020-04-21:
If we do end up adding a `ExternallyAppliedSpatialForceMux`, then I'd advocate for it being primarily a [decorator](https://en.wikipedia.org/wiki/Decorator_pattern).

Specifically, the mux would be a system with N inputs each of type `vector<ExternallyAppliedSpatialForce>` and one output of type `vector<ExternallyAppliedSpatialForce>`.  That way, it can be composed / chained without bound, and the force-providing systems won't care if they are talking to the MbP directly or the mux.  It leaves the connections purely in the domain of the Diagram author.

That probably argues that the propeller system should output a 1-element `vector<ExternallyAppliedSpatialForce>` as well, instead of an `ExternallyAppliedSpatialForce` directly.

We imagine having the mux _additionally_ offer `ExternallyAppliedSpatialForce` input ports directly as sugar for systems that do want to output the force directly without wrapping it in a vector.  That means such systems would always have to connect through the mux, and could never be wired directly into the plant for simplicity.  I don't mind that, but I wonder if its worth the extra complexity.  Perhaps looking at other examples of external forces would help decide.

For the mux implementation with homogeneously typed (`std::vector`) ports only, we could imagine enhancing `primitives::Multiplexer` to handle abstract values.  We'd tell it the number of input ports, and a fold (i.e., reduce) binary function that combines the inputs.  I can help illustrate this if we get that far.

RussTedrake, 2020-04-21:
I had not considered the vector in, vector out version.  Makes sense to me.

I was also thinking of writing this as a more generic multiplexer templated on the abstract type.  Probably as a new system (not on top of the existing Multiplexer).  I think i have enough to take a first pass.

RussTedrake, 2020-04-22:
Here is a more specific proposal on how to implement the multiplexer concept, drake-style:
```
/**
  This system concatenates multiple input ports of type
  std::vector<AbstractValue> into a single output of the same type, by
  concatenating the values.  The input to this system directly feeds through to
  its output.

  @system{ValueMultiplexer, 
    @input_port{vector_input1}
    @input_port{...},
    @input_port{vector_inputN},
    @output_port{vector_output}
  }

  @tparam_default_scalar
  @ingroup primitive_systems
  */
template <typename T>
class ValueMultiplexer : public LeafSystem<T> {
 public:
  DRAKE_NO_COPY_NO_MOVE_NO_ASSIGN(ValueMultiplexer)

  /**
   Constructs a %ValueMultiplexer with `input_sizes.size()` input ports, where
   the input port `i` is a `std::vector<AbstractValue>` with `input_sizes[i]`
   elements.  The single output port is a `std::vector<AbstractValue>` with
   `sum(input_sizes)` elements.
   */
  ValueMultiplexer(const AbstractValue& value, std::vector<int> input_sizes);
...
```
WDYT?
Do we agree with the type-erasure workflow (instead of the templated workflow)?
And I presume pre-allocating everything (demanding particular sizes of the inputs) is best?
Should I also support dynamically sized inputs?

jwnimmer-tri, 2020-04-22:
Yes, I think it should use erasure, not templates.

I don't think `std::vector<AbstractValue>` makes sense (or even compiles?).  An `AbstractValue` is not copyable, it would need `copyable_unique_ptr` around it.

I think the mux's inputs are outputs are just `AbstractValue` directly, and we'd use `Value<vector<ExternallyAppliedSpatialForce>>` at runtime.  That matches the erased type used by MbP so could be directly connected to it.

That means the ValueMultiplexer itself can't re-assemble the output vector, since it won't know they are vectors or how to un-erase the types.

I think (in its constructor) it needs to take a functor that performs the Value-specific composition logic, plus characteristic model_values for the ports (both types are the same in the force mux case, but not in general).

```c++
ValueMultiplexer(
    int num_input_ports,
    const AbstractValue& input_model_value,
    const AbstractValue& output_model_value,
    std::function<...> compose,
```

One option for `compose`'s signature would be
```c++
[](const AbstractValue& one_input_abstract, AbstractValue* output_abstract) {
  using Data = vector<ExternallyAppliedSpatialForce>;
  const auto& one_input = one_input_abstact.get_value<Data>();
  auto* output = one_output.get_value<Data>();
  for (const auto& one_force: one_input) {
    output->push_back(one_force);
  }
}
```

This signature assumes that inputs can be incorporated into the output incrementally.

The other functor option would be to pass all of the inputs (as a vector of abstract value pointers, maybe?) in a single call for performance reasons.  However, my first impression is that the first option can be equally fast.  I _think_ all of the allocations will end up being reused after the first time through the Calc pipeline, so things like `reserve()` will not be necessary.

With some template inference trickery, we could probably even offer `compose` to be
```c++
[](const vector<ExternallyAppliedSpatialForce>& one_input,
   vector<ExternallyAppliedSpatialForce>* output) {
  for (const auto& one_force: one_input) {
    output->push_back(one_force);
  }
}
```

... which would also then infer the two model_value's types as well as the `get_value<>` downcast types.

We'll need the fully-abstract `compose` for pydrake, but perhaps the second form would be useful C++ sugar.  I think could be added later on, though.

RussTedrake, 2020-04-22:
Thanks.  Perhaps for my first pass, I will do `Propeller` => `Propellers` and have a single system output multiple spatial forces.  That's actually makes the simple case -- where i just want to add a bunch of propellers and no other external forces -- quite bit simpler (in addition to skipping the multiplexer, I would have a single `BasicVector<T>` that combines all of the propeller commands in a single vector).

But I think the `Multiplexer` is needed in the longer run.  Perhaps a nice project for dynamics?

amcastro-tri, 2020-04-22:
Not sure why Russ's proposal would not work @jwnimmer-tri. PassThrough has a similar API like:
```c++
explicit PassThrough(const AbstractValue& abstract_model_value);
```
So wouldn't we need to do to `Multiplexer` the same we did with `PassThrough`?

jwnimmer-tri, 2020-04-22:
> ... Not sure why Russ's proposal would not work  ...

It's because ...

> This system concatenates multiple input ports of type `std::vector<AbstractValue>` into a single output of the same type ...

... a `std::vector<AbstractValue>` does not compile:

https://github.com/RobotLocomotion/drake/blob/2c52ec67bc3e3579b9b10f65a58116c212889cec/common/value.h#L64-L66

amcastro-tri, 2020-04-22:
sry if asking a dumb question.
Could it be possible that `T` in `class ValueMultiplexer` is the actual type? as we do with the `Value` class? Then that'd allow us to say `ValueMultiplexer<ExternallyAppliedSpatialForce<double>>` or similar and then we'd do `std::vector<T>`. I'm sure I'm missing something....

jwnimmer-tri, 2020-04-22:
It's a fine question.  That's what Russ was asking when he said "Do we agree with the type-erasure workflow (instead of the templated workflow)?".

The templated workflow would mean that it'd be almost impossible to incorporate the ValueMultiplexer into pydrake.  For a reusable primitive system, that's a showstopper for me.

amcastro-tri, 2020-04-22:
> That's what Russ was asking when he said "Do we agree with the type-erasure workflow (instead of the templated workflow)?".

Ah! slow me, sry. Most of the trickery to write those bindings is still foreign to me. I can see the issue though. Thanks for the answer.

jwnimmer-tri, 2020-04-22:
To a first approximation, the way to think about templates and pydrake is to ask whether the allowed set of template arguments is open or closed.  For T as a scalar type, it's closed (specified in the class docs as `@tparam_nonsymbolic_scalar` or whatever.)  For your multiplexer proposal, it's open (any data type the user could conceive of).  Binding the closed set is easy (we just loop over the list of types).  Binding an open set of types at compile time is extremely cumbersome.  That's where type erasure helps.

sherm1, 2020-04-22:
Hate to throw a wet blanket over the excitement, but how about
`ExternalSpatialForceMultiplexer<T>` as the plant? No templates, no type erasure! Takes any number of `vector<ExternallyAppliedSpatialForce<T>>` input ports and provides one `vector<ExternallyAppliedSpatialForce<T>>` output port.

Current use case solved, future use cases punted to the future!

PR #13135 (2020-04-25): multibody: adds a Propeller model for use with MultibodyPlant
Includes python bindings.
Partially resolves #12401.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13135)
commit 0ab38013 (2020-04-25): multibody: adds a Propeller model for use with MultibodyPlant (#13135)

* multibody: adds a Propeller model for use with MultibodyPlant

Includes python bindings.
Partially resolves #12401.
closed 2020-04-25 by RussTedrake