RobotLocomotion/drake #12420: AddSosConstraint bug

akshitj1, 2019-11-27:
Hi, I was trying to find region of attraction for time varying LQR controller on pendulum swing up trajectory. For SOS feasibility program eq. (39) in this [paper](https://groups.csail.mit.edu/robotics-center/public_papers/Tedrake10.pdf) I need to use `AddSosConstraint` [here](https://github.com/akshitj1/drake/blob/026e3581ff60ce11c8bef3e60780b730230f568e/examples/pendulum/funnel_library/simulate.cc#L57). Upon calling `Solve`, program terminates with error `Constraint 1 is empty. exited with status = 206`. Upon debugging I found that `AddSosConstraint` forms sos_polynomial which has total degree less than constraint polynomial degree (ref `poly_diff` in [this call](https://github.com/RobotLocomotion/drake/blob/master/solvers/mathematical_program.cc#L1096)), hence it adds linear constraints on coefficients with no decision variables(eg. 0.5*x^3 -> 0.5==0). 
Easy steps to reproduce:
```
    MathematicalProgram prog;
    const VectorX<Variable> vars{prog.NewIndeterminates<2>(
        std::array<std::string, 2>{"x", "y"})};

    Polynomial p;
    // p = x + x*y + x*y^2
    for (size_t i = 0; i < 3; i++) {
      std::map<Variable, int> terms = {{vars(0), 1},
                                       {vars(1), i}};
      p.AddProduct(1.0, symbolic::Monomial(terms));
    }
    prog.AddSosConstraint(p);
    solvers::Solve(prog);
```
In this polynomial exponents are:
```
[ 1 0;
  1 1;
  1 2 ]
```
But monomial basis is empty.
Here it is easy to debug and I can correct the constraint but it the actual case I am unable to find the problem. I tried to find bug in `ConstructMonomialBasis` but can't understand `RemoveWithRandomSeparatingHyperplanes` part and monomials are being removed there. Can you please help.

hongkai-dai, 2019-11-27:
What happens here is that when computing the monomial basis, we know that the monomial basis should be within the Newton polytope
```
floor ( 0.5 * [1 0; 1 1; 1 2])
= [0 0; 0 0; 0 1]
```
hence the monomial basis only contains a single term `y`. (For more details on how to determine the monomial basis of a sos polynomial through Newton Polytope, you could refer to section 3.3.4 of https://sites.math.washington.edu/~thomas/frg/frgbook/SIAMBookFinalvNov12-2012.pdf)

We then tried to add the following constraint to the program when calling `AddSosConstraint`
```
y * S * y = x + x*y + x*y^2
S >= 0
```
By comparing the coefficients of `y * S * y = x + x*y + x*y^2`, we get the following linear equality constraints
```
1 = 0 (coefficient for x)
1 = 0 (coefficient for x * y)
S = 0 (coefficient for y * y)
1 = 0 (coefficient for x * y^2)
```
Adding this type of constraint (1 = 0) to Mosek causes issue, hence you see the error. 

We can try to throw more meaningful errors. But I would suggest to look at your polynomial that you want to be sum-of-squares. Apparently the polynomial (x + x*y + x*y^2) **cannot** be sos, as the highest order term x*y^2 is an odd degree monomial, and sos requires even degree monomial as the highest order term.

akshitj1, 2019-11-27:
Thanks @hongkai-dai for the explanation. I didn't assumed given example to be SOS as commented "Here it is easy to debug and I can correct the constraint but it the actual case I am unable to find the problem". Above was simply to compactly reproduce as my original error. My original SOS constraint(pendulum swing up SOS case) was rather long polynomial in total degree 8 and contained lot of terms, So was unable to find problem. I thought `Solve` method will result to infeasible instead of exiting with missing constraint error. I increased the free polynomials degree after reading your explanation. Now solver works fine. But yes, better assertions will help in early and meaningful failure.

hongkai-dai, 2019-11-27:
I agree it is better to detect the problem earlier inside Drake, rather than Mosek solver throwing some error. I will work on that separately. Thanks!

closed 2019-11-27 by hongkai-dai