RobotLocomotion/drake #12441: Hydroelastic contact surface broken for thin rigid object -- needs to use normals

sherm1, 2019-12-05:
A typical hydroelastic contact for us is a soft gripper (say a bubble) gripping a thin rigid object (a plate, a spoon, a thin-walled coffee cup). In that case the real-life deformation of the gripper can easily (and reasonably) exceed the thickness of the object. In hydroelastics we always work with the _undeformed_ geometry (because the pressure field models the deformation). So the situation looks something like this:
[image]

Our current algorithm produces contact surfaces on both sides of the "embedded" spoon, with consequent forces on both sides that nearly cancel. In real life of course there is only pressure on the bottom surface of the spoon because the deformed gripper would only be on that side. That is way wrong!

I believe there is an easy fix: we should not include contact surface triangles for rigid/soft contact when the rigid object surface normal opposes the pressure field gradient. That's easy to check while we're collecting the contact surface triangles.

@DamrongGuoy @edrumwri @ryanelandt @SeanCurtis-TRI

SeanCurtis-TRI, 2019-12-05:
So, this provides sufficient motivation to include pre-computation of the per-tet, constant gradient value for soft meshes even if we're not doing soft-soft contact. So, that's going to have to get moved up in our timeline.

DamrongGuoy, 2019-12-05:
Using soft pressure gradients and rigid surface normal is an interesting idea.  It works in the shown example, even though there are other examples that doesn't work.

In terms of implementation, it shouldn't be hard to filter out the unwanted parts.

SeanCurtis-TRI, 2019-12-05:
@DamrongGuoy and I chatted and would provide an argument for turning the knob up to 11.

Consider this case:

[image]

If we eliminate the "back facing" surface we'd eliminate this blue region:

[image]

However, if we ask, how did we get in this configuration, we probably need to cull more. Imagine that the soft region is a half space. Which means we came in from above. Because we came in from above, the leading edge should get 100% of the pressure. The trailing surfaces are in the lee of that surface and shouldn't be reporting any pressure. (i.e., we should be eliminating both the red and blue surfaces in the next image).

[image]

Of course, this could also introduce a question of history. If the soft thing is *not* a half space, but a box, it's possible that we came in from the left side of the screen. At that initial contact, all of the surface was originally a leading front. And what we've done is we've pushed *past* that domain where the whole surface mattered, into a regime where regions of the contact surface would be considered non-contributory.

ryanelandt, 2019-12-05:
Trying to create a general, deformation-free model of deformation is not practical for situations like this. I can think of four possible modeling solutions (I'm likely missing some) that apply for certain situations. I'll try to outline what each is and the advantages and disadvantages.

1. a

ryanelandt, 2019-12-05:
There is no single tweak that makes PFC generally-well behaved for situations like this. I can think of three possible modeling solutions (I'm likely missing some) that apply for certain situations. I'll try to outline what each is and the advantages and disadvantages.

1. You ignore the problem. I did this for my spoon video a year ago and didn't notice anything visually wrong. (The bar set by point-contact is really very low.) If the amount of foam on the "wrong side" of the spoon is small relative to the spoon thickness controllers designed may still work fine. You can do other tweaks including ignoring damping and friction when gradients point in same direction (this needs to be done continuously) to improve the effectivness of this work-around and still gaurantee physical behavior in general.

2. You modify pressure with a continuous function that sets pressure to zero when the pressure field gradients point in the same direction (dot product of gradients is positive). Doing this creates a non-conservative system and opens you up to an energy exploits if used in optimization applications. Energetically, it should be better-behaved for geometries with near-constant projected areas (e.g., sphere) and poorly behaved for long and thin obejects. Note: the energy created/destroyed can be tracked in needed.

3. You modify the geometry. If you plan to only grip the coffee cup on the outside you can model the cup as solid. Sean's pathalogical example can likely be modeled with it's convex hull. Get creative, an object can have multiple pressure fields and use one for contacts with certain objects or grasps. For example, a cup could be solid when gripping it on the outside, but hollow when trying to pinch it.

sherm1, 2019-12-05:
Because this is a serious limitation for real applications we should at least mention it in the paper. For TRI's applications (e.g. soft gripper picks up spoon, plate, thin-walled coffee mug, etc.) I think removing the badly-oriented faces is a reasonable fix, and we will have to do _something_ since these are our bread-and-butter cases.

sherm1, 2019-12-05:
Let's discuss on Wednesday.

DamrongGuoy, 2020-04-01:
Now we have an intermediate solution in master.  It checks the angle between the rigid geometry's surface normal and the soft geometry's pressure gradient vector.

I have an example in my branch https://github.com/DamrongGuoy/drake/tree/contact_thin_objects `bazel run //examples/scene_graph:contact_surface_thin_object`. It has a moving soft ball between a rigid thin plate and a rigid bowl.  It shows only one surface instead of two surfaces on both sides of the thin plate and the bowl. I attached two movies of that example in a ZIP file.

This intermediate solution has limitation, and I'll create another issue for a better solution in the future.

[plate_ball_bowl_movies.zip](https://github.com/RobotLocomotion/drake/files/4412149/plate_ball_bowl_movies.zip)

closed 2020-04-01 by DamrongGuoy