RobotLocomotion/drake #12445: Suggested improvement for `mkdocs.py`

tvatter, 2019-12-05:
A suggestion to improve: https://github.com/RobotLocomotion/drake/blob/master/third_party/com_github_pybind_pybind11/mkdoc.py#L275

```
    # Convert `typewriter` to ``typewriter``.
    rr = r'([\s\-,;:!.()]+)`([^\s`]|[^\s`].*?[^\s`])`([\s\-,;:!.()]+)'
    while re.search(rr, s):
        s = re.sub(rr, r'\1``\2``\3', s, flags=re.DOTALL)
```

(I noticed that the current version only converts the first `typewriter`, at least on my ubuntu machine)

tvatter, 2019-12-05:
Another one for https://github.com/RobotLocomotion/drake/blob/master/third_party/com_github_pybind_pybind11/mkdoc.py#L245

```
if s.startswith('///<'):
    s = s[4:]
```

(this should come right before, it's useful for enum classes)

m-chaturvedi, 2020-01-02:
@tvatter:
https://repl.it/@MmanuChaturvedi/BouncyWrithingState

Reference: https://docs.python.org/3/library/re.html#re.sub
In the default case, `count=0`, it should replace all the occurrences.

Were you using anything from `[\s\-,;:!.()]+` at the start and end?

PR #12519 (2020-01-03): mkdoc: Fix comment handling edge cases
Closes #12445

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12519)
closed 2020-01-03 by EricCousineau-TRI