RobotLocomotion/drake #12453: Memory leak from SpMat<Expression> * SpMat<Expression>

soonho-tri, 2019-12-06:
Reproducible step:
```c++
   Eigen::SparseMatrix<Expression> Q1(2, 2);
   Eigen::SparseMatrix<Expression> Q2(2, 2);
   (Q1 * Q2).eval();    // <-- HERE
```

This is due to this line in Eigen codebase: https://bitbucket.org/eigen/eigen/src/cb61fd12032cb7dcd7706aa201bfcec0a0e7b06a/Eigen/src/Core/ProductEvaluators.h#lines-110

Here is a small program illustrating the issue:

```c++
#include <iostream>
#include <memory>

using std::cerr;

// Simplified Implementation of drake::symbolic::Expression
class MyScalar {
public:
  MyScalar(const double v) : ptr_{std::make_shared<double>(v)} {
    cerr << "MyScalar(double)\n";
  }
  ~MyScalar() { cerr << "~MyScalar()\n"; }

private:
  std::shared_ptr<double> ptr_;
};

// Simplified version of the specialization of `Eigen::internal::evaluator` at
// https://bitbucket.org/eigen/eigen/src/94875feeeeb9abe5509b314197da1991ba2070f5/Eigen/src/SparseCore/SparseCompressedBase.h#lines-308:364.
struct evaluator {
  evaluator() : zero_{0.0} { cerr << "evaluator()\n"; }
  ~evaluator() { cerr << "~evaluator()\n"; }

private:
  const MyScalar zero_;
};

// Simplified version of the specialization of
// `Eigen::internal::product_evaluator` at
// https://bitbucket.org/eigen/eigen/src/94875feeeeb9abe5509b314197da1991ba2070f5/Eigen/src/Core/ProductEvaluators.h#lines-93:129.
struct product_evaluator : public evaluator {
  product_evaluator() {
    cerr << "product_evaluator()\n";
    ::new (static_cast<evaluator *>(this)) evaluator();
  }
  ~product_evaluator() {
    // static_cast<evaluator *>(this)->~evaluator();
    cerr << "~product_evaluator()\n";
  }
};

int main() { product_evaluator e; }
```

GodBolt Link: https://godbolt.org/z/ztZa7B

At line 34, it explicitly calls the constructor of `evaluator`. So the constructor of `evaluator` is called twice. But its destructor is called only once.

jwnimmer-tri, 2019-12-07:
It seems like Eigen is presuming that scalars are trivially default constructible?  If course it should be using Traits and not assume that.

However, in the meantime -- at the cost of promoting most of our `DRAKE_ASSERT(e.ptr_ != nullptr)` to if-guards instead -- we could probably leave Expression's `std::shared_ptr<ExpressionCell> ptr_;` as nullptr when default constructed, and perhaps even alter `Expression(double)` to leave the cell as nullptr iff the argument is 0.0.

In that case, we have (1) the memset-to-zero representation and default-constructed representation and construct-with-zero representation are identical in memory, and (2) repeatedly using placement-new or default constructor or construct-with-zero on the same memory are idempotent.  (The binary representation of a null shared_ptr happens to be all zeros.)  This might be enough to paper over Eigen's bug here.

Using a `nullptr` cell for zero might even end up faster, since we would not have to keep the refcount updated on the ExpressionCell for Zero anymore when allocating large, empty matrices?

jwnimmer-tri, 2019-12-07:
Here's another crazy idea.  We could change the representation of `Expression` to be something like `std::variant<double, std::shared_ptr<ExpressionCell>>` and get rid of `ExpressionConstant`.  With a few tricks, I bet we could even do it without using more memory, and with the representation of `0.0` being bitwise zero.

soonho-tri, 2019-12-07:
My current work-around is to make sure that all of `Expression(0.0)` uses the cell of `Expression::Zero()`. In this way, we can avoid memory-allocation in case of `Expression(0.0)`.

I will think more about the alternatives that you suggested above. Thanks for the thoughts!

jwnimmer-tri, 2019-12-07:
Ah -- and because we never need to destroy the singleton ExpressionCell for Zero, we don't care if its refcount becomes artificially inflated by unmatched placement new.  Nifty.

soonho-tri, 2019-12-07:
Yup, it was the simplest solution that I can think of. I am thinking about "patching" it in this way first.

In the future, I will re-implement the symbolic classes using `std::variant`. The use of`Cell` pattern in the design was to mimic variant types, which now we have.

soonho-tri, 2019-12-09:
Close by #12459

closed 2019-12-09 by soonho-tri