RobotLocomotion/drake #12454: Flipping contact surface normal convention caused contact reporting to break

edrumwri, 2019-12-06:
PR #12432 made the usage of the contact surface normals consistent with the latest version of the hydroelastic paper (normals point from geometry N (body B) into geometry M (body A)). As a result, the spatial force for reporting gets changed from "applied to body A" to "applied to body B", though the hydroelastic contact reporting specifies that the spatial force it is constructed with should be applied to body A.

This means that the visualization of the spatial force for hydroelastics (and any other present usages of the hydroelastic spatial force from reporting) is currently wrong. 

My personal preference would be to change the convention so that the normal points from Body B into Body A (which matches the literature on contact that I was raised on). @SeanCurtis-TRI points out that this would also make the definition of the normal consistent with what we use for point pair contacts.

cc @sherm1 @amcastro-tri @SeanCurtis-TRI @DamrongGuoy

SeanCurtis-TRI, 2019-12-06:
First sentence is in error: #12432 made the normals *from M (body A) into N (body B)*. See here:

https://github.com/RobotLocomotion/drake/blob/master/geometry/query_results/contact_surface.h#L99-L102

SeanCurtis-TRI, 2019-12-06:
Having the normal point from body B and into body A (or from geometry M and into geometry N) would make it consistent with both the `PenetrationAsPointPair` but also the `SignedDistancePair` where we consistently exercise that convention.

It would make it different from the most recent version of the paper.

edrumwri, 2019-12-06:
> First sentence is in error: #12432 made the normals _from M (body A) into N (body B)_. See here:
> 
> https://github.com/RobotLocomotion/drake/blob/master/geometry/query_results/contact_surface.h#L99-L102

Edited the text of the issue to reflect this.

DamrongGuoy, 2019-12-06:
Thank you for letting me know. It is nice to have consistent normal convention before I start working on filtering out unwanted parts of the intersection between rigid surface and soft volume (#12441), probably a few weeks from now.

edrumwri, 2019-12-09:
Per DrakeDeveloper's meeting: the solution we've decided on is changing the normal convention in hydroelastics to point from Body B to Body A.

sherm1, 2019-12-09:
(The B->A convention is what Drake is already using for non-hydroelastic contact queries. Evan says most contact literature also uses the B->A convention -- our PFC paper is an outlier.)

PR #12463 (2019-12-12): Reverses the surface normal convention in ContactSurface.
Resolves #12454.

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12463)
closed 2019-12-12 by edrumwri