RobotLocomotion/drake #12456: Compile error in SparseMatrix<double> * SparseMatrix<Expression>

soonho-tri, 2019-12-07:
How to reproduce:
```c++
  Eigen::SparseMatrix<double> M1(2, 2);
  Eigen::SparseMatrix<Expression> M2(2, 2);
  (M1 * M2).eval();  //  <-- HERE
```

Error message:
```
external/eigen/Eigen/src/SparseCore/ConservativeSparseSparseProduct.h:54:14: error: no viable conversion from 'const Eigen::SparseCompressedBase<Eigen::SparseMatrix<drake::symbolic::Expression, 0, int> >::Scalar' (aka 'const drake::symbolic::Expression') to 'Scalar' (aka 'double')
      Scalar y = rhsIt.value();
             ^   ~~~~~~~~~~~~~
external/eigen/Eigen/src/SparseCore/ConservativeSparseSparseProduct.h:150:17: note: in instantiation of function template specialization 'Eigen::internal::conservative_sparse_sparse_product_impl<Eigen::SparseMatrix<double, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int> >' requested here
      internal::conservative_sparse_sparse_product_impl<Lhs,Rhs,ColMajorMatrix>(lhs, rhs, resCol, true);
                ^
external/eigen/Eigen/src/SparseCore/SparseProduct.h:76:104: note: in instantiation of member function 'Eigen::internal::conservative_sparse_sparse_product_selector<Eigen::SparseMatrix<double, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, 0, 0, 0>::run' requested here
                                                          typename remove_all<RhsNested>::type, Dest>::run(lhsNested,rhsNested,dst);
                                                                                                       ^
external/eigen/Eigen/src/SparseCore/SparseProduct.h:43:5: note: in instantiation of function template specialization 'Eigen::internal::generic_product_impl<Eigen::SparseMatrix<double, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, Eigen::SparseShape, Eigen::SparseShape, 8>::evalTo<Eigen::SparseMatrix<drake::symbolic::Expression, 0, int> >' requested here
    evalTo(dst, lhs, rhs, typename evaluator_traits<Dest>::Shape());
    ^
external/eigen/Eigen/src/Core/ProductEvaluators.h:124:69: note: in instantiation of function template specialization 'Eigen::internal::generic_product_impl<Eigen::SparseMatrix<double, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, Eigen::SparseShape, Eigen::SparseShape, 8>::evalTo<Eigen::SparseMatrix<drake::symbolic::Expression, 0, int> >' requested here
    generic_product_impl<Lhs, Rhs, LhsShape, RhsShape, ProductTag>::evalTo(m_result, xpr.lhs(), xpr.rhs());
                                                                    ^
external/eigen/Eigen/src/Core/ProductEvaluators.h:35:62: note: in instantiation of member function 'Eigen::internal::product_evaluator<Eigen::Product<Eigen::SparseMatrix<double, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, 2>, 8, Eigen::SparseShape, Eigen::SparseShape, double, drake::symbolic::Expression>::product_evaluator' requested here
  EIGEN_DEVICE_FUNC explicit evaluator(const XprType& xpr) : Base(xpr) {}
                                                             ^
external/eigen/Eigen/src/SparseCore/SparseAssign.h:77:20: note: (skipping 2 contexts in backtrace; use -ftemplate-backtrace-limit=0 to see all)
  SrcEvaluatorType srcEvaluator(src);
                   ^
external/eigen/Eigen/src/SparseCore/SparseAssign.h:38:13: note: in instantiation of member function 'Eigen::internal::Assignment<Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, Eigen::Product<Eigen::SparseMatrix<double, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, 2>, Eigen::internal::assign_op<drake::symbolic::Expression, drake::symbolic::Expression>, Eigen::internal::Sparse2Sparse, void>::run' requested here
          ::run(derived(), other.derived(), internal::assign_op<Scalar,typename OtherDerived::Scalar>());
            ^
external/eigen/Eigen/src/SparseCore/SparseMatrix.h:1124:18: note: in instantiation of function template specialization 'Eigen::SparseMatrixBase<Eigen::SparseMatrix<drake::symbolic::Expression, 0, int> >::operator=<Eigen::Product<Eigen::SparseMatrix<double, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, 2> >' requested here
    return Base::operator=(other.derived());
                 ^
external/eigen/Eigen/src/SparseCore/SparseMatrix.h:684:15: note: in instantiation of function template specialization 'Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>::operator=<Eigen::Product<Eigen::SparseMatrix<double, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, 2> >' requested here
        *this = other.derived();
              ^
external/eigen/Eigen/src/SparseCore/SparseMatrixBase.h:385:14: note: in instantiation of function template specialization 'Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>::SparseMatrix<Eigen::Product<Eigen::SparseMatrix<double, 0, int>, Eigen::SparseMatrix<drake::symbolic::Expression, 0, int>, 2> >' requested here
    { return typename internal::eval<Derived>::type(derived()); }
```

1. Eigen-3.3.3 (the version that we're using) has a bug. See https://gitlab.com/libeigen/eigen/blob/3.3.3/Eigen%2Fsrc%2FSparseCore%2FConservativeSparseSparseProduct.h#L54.  `Scalar` (`double`) is used to hold a symbolic expression (`
`rhsIt.value();`) and causes a compile error.

2. Eigen-3.3.5 fixes the problem by using two distinct types, `LhsScalar` and `RhsScalar`. See https://gitlab.com/libeigen/eigen/blob/3.3.5/Eigen%2Fsrc%2FSparseCore%2FConservativeSparseSparseProduct.h#L17-57.

3. I am afraid that https://github.com/RobotLocomotion/drake/issues/11758 is not going to help because bionic provides libeigen-3.3.4 which still has this issue. See https://gitlab.com/libeigen/eigen/blob/3.3.4/Eigen%2Fsrc%2FSparseCore%2FConservativeSparseSparseProduct.h#L54.

4. Note that the other case, `SparseMatrix<Expression> * SparseMatrix<double>`, has no problem. Since an `Expression` object can hold a `double` value.

jwnimmer-tri, 2019-12-07:
The work-around is `(M1.template cast<Expression>() * M2).eval()`, I think?

soonho-tri, 2019-12-07:
Certainly, that should work. I can work on providing `operator*` overloading for the combinations of {Expression, Variable, double} for Eigen::SparseMatrix.

At the same time, I'd like to know if we can use eigen-3.3.5 or eigen-3.3.7.

jwnimmer-tri, 2019-12-07:
To answer that, I would like to understand if 3.3.7 is ABI-compatible with the host-provided Eigen (including the `unsupported/AutoDiff` module), and I haven't looked into that.  If work-arounds are not too much effort, the boring answer of "use the host Eigen" is still my starting point.

jwnimmer-tri, 2021-11-18:
On macOS, we use Eigen 3.4.  On Focal, we use Eigen 3.3.7.  According to the above, both of these already succeed.

On Bionic, we have Eigen 3.3.4 which has this problem, but per #13391 we will be dropping that support about four months from now.

Therefore, closing this as "won't fix".

closed 2021-11-18 by jwnimmer-tri