RobotLocomotion/drake #12460: SetFixedInputPortValue()

KarimHassanieh, 2019-12-07:
Hi ! Im new here and trying to get the hang of drake I wrote the below code for the simulation of simple pendulum but im having difficulty with putting the input port value I keep getting the below error : "Failure at systems/framework/context_base.cc:85 in SetFixedInputPortValue(): condition '0 <= index && index < num_input_ports()' failed"
Its not clear to me exactly how to resolve the error and would appreciate any help in this matter if possible ..I attached the code here if anyone needs to look at it as well
[code.txt](https://github.com/RobotLocomotion/drake/files/3935614/code.txt)

kindest regards
Karim

sherm1, 2019-12-08:
Hi, Karim. Looks to me like you are just missing a call to DiagramBuilder's [ExportInput()](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_diagram_builder.html#a1db762bc34903d5e5eb8f1c26ec2f255) method to expose the pendulum system's input port to the top-level diagram. Otherwise the Diagram doesn't have any input ports, which would explain the error message you are seeing. If you look for "ExportInput" in Drake examples you'll see how it is used.

Also, I happened to notice a bug in your `DoCalcTimeDerivatives()` implementation. You have
```c++
double theta=context.get_continuous_state()[0];
double theta_dot=context.get_continuous_state()[0];
```
The second index should be a 1.

Please try these fixes, post whether that works, and if so close the issue.

BTW, the preferred method for getting Drake help is to ask a Stackoverflow question with tag "drake" rather than filing an issue (many people get spam when you file an issue). Please see our [Getting Help](https://drake.mit.edu/getting_help.html) page.

KarimHassanieh, 2019-12-08:
Thanks it worked! Also noted for using Stackoverflow..thanks for the help!

closed 2019-12-08 by KarimHassanieh