RobotLocomotion/drake #12468: Support shape permutations across scalar types in ComputeSignedDistancePairwiseClosestPoints

tehbelinda, 2019-12-10:
Following up from #10153, there is still more work required for full proximity engine support. Currently only certain shape permutations are supported and only with certain scalars. Even for doubles, there are varying levels of precision when falling back to using FCL. In `distance_to_shape_callback.h`, the current supported configurations are:

|                   | Sphere     | Box                         | Cylinder               | Halfspace              | Capsule              |
| -------------- | ------------- | ------------------------- | ---------------------- |---------------------- |---------------------- |
| **Sphere** | double<br>Autodiff | double<br>Autodiff | double | double<br>Autodiff | double (FCL fallback)                      |
| **Box**      | double      | double (FCL fallback)       | double (FCL fallback) | :x:                       | double (FCL fallback) |
| **Cylinder**  | double   | double (FCL fallback)       | double (FCL fallback) | :x:                       | double (FCL fallback) |
| **Halfspace** | double | :x:              | :x:        | :x:                       | :x:                       |
| **Capsule**   | double (FCL fallback)  | double (FCL fallback)       | double (FCL fallback) | :x:                       | double (FCL fallback)                | 

We should fully support the shapes we care about, i.e. shapes with spheres, by providing our own implementation for sphere-capsule in `distance_to_point_callback.h`. Otherwise we should provide fallbacks and update the documentation to reflect the latest state.

cc @SeanCurtis-TRI @DamrongGuoy

SeanCurtis-TRI, 2019-12-10:
Ideally, we'd like to take this one step further. Let's include the level of precision actually provided in the table. It should certainly be included when this table ends up in the documentation.

DamrongGuoy, 2020-10-06:
This issue #12468 is similar to #10907.  May I add the content of this issue into #10907 and close this issue?  Or we prefer to keep them apart and fix one issue at a time?

SeanCurtis-TRI, 2020-10-06:
I like rolling them in together.  Good catch!

DamrongGuoy, 2020-10-06:
Thank you.  I close this issue in favor of #10907.

closed 2020-10-06 by DamrongGuoy