RobotLocomotion/drake #12473: Welding a small mass/inertia link to an existing parent causes MBP's `CalcReactionForces` to fail in debug mode

rcory, 2019-12-12:
As demonstrated in PR #12469, welding an infinitesimally small mass/inertia (mass=ixx=iyy=izz=1e-7) to an existing link (with proper mass/inertia values) via a `WeldJoint` causes [this](https://github.com/rcory/drake/blob/8e43282492c9f7f3dcd4370b1e6b4701118d31e1/multibody/plant/multibody_plant.cc#L2686-#L2687) assert statement in `MultibodyPlant::CalcReactionForces` to fail in debug mode.

amcastro-tri, 2019-12-12:
Thank you @rcory. Rick provided a branch that reproduces this problem [here](https://github.com/rcory/drake/tree/debug_gripper_lcm) (hash f48f404461381316504a983732ea3f33a7ada236).

As we discussed f2f yesterday, a possible cause could be that we use ID to compute reaction forces *after* we already updated the context with TAMSI. That includes both configuration terms (mass matrix will not be M(q0) but M(q) and velocity terms, like gyroscopic forces).

amcastro-tri, 2019-12-12:
I studied the code in more detail. Yes, my suspicion is correct. We are computing reaction forces with an updated context while TAMSI lags some terms behind (for instance M and C). 
I believe the quickest solution would be to cache the quantities that TAMSI computed so that we can re-use them when computing reaction forces. 
If this is correct, we should see a dramatic drop (to machine epsilon) in the residual of the computation.

Also, regarding your PR @rcory, commenting out this check is not a bad solution after all. The errors introduced by updating the state are definitely not within machine epsilon, but they are O(h) (time step) and thus reaction forces still are reliable.

sherm1, 2020-04-02:
@amcastro-tri is this still an issue?

amcastro-tri, 2020-04-02:
The assert was removed in #12469. We were asking for a residual close to machine epsilon while the analysis above reveals the error is O(h) (with h time step). Still a good estimation of reaction forces with order of accuracy consistent with the method's accuracy.

Therefore I will close for now unless we believe we want to drop the residual to machine epsilon for some other reason.

closed 2020-04-02 by amcastro-tri

xuchenhan-tri, 2022-05-25:
In MbP.cc, there is

```
  // Since vdot is the result of Fapplied and tau_applied we expect the result
  // from inverse dynamics to be zero.
  // TODO(amcastro-tri): find a better estimation for this bound. For instance,
  // we can make an estimation based on the trace of the mass matrix (Jain 2011,
  // Eq. 4.21). For now we only ASSERT though with a better estimation we could
  // promote this to a DEMAND.
  // TODO(amcastro-tri) Uncomment this line once issue #12473 is resolved.
  // DRAKE_ASSERT(tau_id.norm() <
  //              100 * num_velocities() *
  //              std::numeric_limits<double>::epsilon());
```
Should we turn on the assert here? @amcastro-tri