RobotLocomotion/drake #12483: docker: Unable to launch drake-visualizer?

EricCousineau-TRI, 2019-12-14:
Not sure if this was intended to be a supported workflow, but a user cannot launch `drake-visualizer` from the Bazel-build Docker container. I can reproduce this issue on our Dockerhub images as well.

https://stackoverflow.com/questions/59330890/drake-libgl-error-when-starting-visualizer-for-the-first-time-through-docker/59331832#59331832

Is this something that we can fix easily?

EricCousineau-TRI, 2019-12-14:
(I thought that the non-NVidia docker image was able to run Director with software GL... Am I mis-remembering?)

jamiesnape, 2019-12-16:
This relates to the age old question of whether any of the build-your-own Docker images are supported. Getting GL working is even more problematic and as it partly depends on the host machine's capabilities too.

I could probably fix it for this user, but in general, I would rather nuke the instructions as I (since it is pretty much always myself) can't support all the combinations of configurations. At best, I can support Jupyter and Meshcat.

I know we are not here, but if we were talking about the Docker Hub images, that would be a different question, of course.

EricCousineau-TRI, 2019-12-16:
Actually, I'd be more than happy for us to constrain this issue to the DockerHub images, as that's where I was able to reproduce this:
https://stackoverflow.com/a/59331832/7829525

jamiesnape, 2019-12-16:
I don't think we intended the Docker Hub images to work with `drake-visualizer`, so I don't doubt that.

jwnimmer-tri, 2020-05-14:
> This relates to the age old question of whether any of the build-your-own Docker images are supported. 

Per #7654.

> Is this something that we can fix easily?

Sounds like empirically no, even for just dockerhub?
Would any of the other visualizers work more easily?

jamiesnape, 2020-05-14:
> Sounds like empirically no, even for just dockerhub?

The problem is matching the host configuration with the guest, so the Docker Hub image would not be easy to fix.

> Would any of the other visualizers work more easily?

Any visualizer that is actually running outside of the Docker container and working with data sent from inside the Docker container is going to be the solution here. So to an extent you could make a host drake-visualizer work, but there are easier solutions that would have a more lightweight client.

jwnimmer-tri, 2020-05-28:
My reading of the above thread is that this issue is "WONTFIX".  Per #7654 we will remove the build-your-own Docker, and making Docker Hub work is not-trivial.  We should focus on other visualizers in this context, I believe.

closed 2020-05-28 by jwnimmer-tri

jamiesnape, 2020-05-29:
Should I remove `drake-visualizer` from the prebuilt Docker images to avoid future disappointment and support requests.

EricCousineau-TRI, 2020-05-29:
I don't really support the idea of actively removing components from a simple deployment process, so my opinion is no. But that's just me.

jamiesnape, 2020-05-29:
If it is broken, why keep it? If we distribute it, we need to support it.

jwnimmer-tri, 2020-05-29:
Instead of removing files, I think failing faster or providing a diagnostic would be superior.  Removing a file leads to potential user confusion about why it is missing.

The awkward version of that would be to rename `drake-visualizer` to `drake-visualizer.image` and add `drake-visualizer` as a stub that just prints a cautionary note with a citation to the renamed file in case the user is feeling bold.

The spectacular version of that would be if the visualizer (or a wrapper) self-diagnosed whether the user's graphics stack was hosed, and pointed them to advice on what to do about it.

weigewansui, 2021-02-22:
I'm able to get this working in container on Ubuntu base image.
basically you have to:
1. make sure your NVIDIA driver is running correctly in container using Nvidia's docker runtime.
2. Run in previlliged mode
3. Mount all the necessary dirs for GUIs  "-e DISPLAY=$DISPLAY -e QT_X11_NO_MITSHM=1 -v $HOME/.Xauthority:/root/.Xauthority -v /tmp/.X11-unix:/tmp/.X11-unix --privileged "

PR #13764 (2020-08-10): Replace drake-visualizer with warning script on prebuilt Docker images
On prebuilt Docker images, rename `drake-visualizer` to
`drake-visualizer.image`. Create a `drake-visualizer` script that prints a
warning message with a pointer to the renamed file.

Resolves #12483 and #13639

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13764)
commit fc3e6be8 (2020-08-10): Replace drake-visualizer with warning script on prebuilt Docker images (#13764)

* Replace drake-visualizer with warning script on prebuilt Docker images

On prebuilt Docker images, rename `drake-visualizer` to
`drake-visualizer.image`. Create a `drake-visualizer` script that prints a
warning message with a pointer to the renamed file.

Resolves #12483 and #13639