RobotLocomotion/drake #12484: the format of argument of pydrake.forwarddiff.jacobian ?

vinbo, 2019-12-14:
the function description:　https://drake.mit.edu/pydrake/pydrake.forwarddiff.html#pydrake.forwarddiff.jacobian

my code: 
`
from pydrake.forwarddiff import jacobian
import math
import numpy as np

def f(t):
    return np.array([1.5 - 0.5 * sigmoid(t - 2.5), 0.5, 0.0])
print jacobian(f,0.)
`
The returned error info: 
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-76474f9ccfc4> in <module>()
     48 
     49 
---> 50 print jacobian(f,0.)
     51 #print jacobian(f,np.array[0.])
     52 

/drake/lib/python2.7/site-packages/pydrake/forwarddiff.py in jacobian(function, x)
     38     The function should be vector-input and vector-output.
     39     """
---> 40     x_ad = np.empty(x.shape, dtype=np.object)
     41     for i in range(x.size):
     42         der = np.zeros(x.size)

AttributeError: 'float' object has no attribute 'shape'

If I changed that line into
print jacobian(f,np.array[0.])

then the error becomes:

The autoreload extension is already loaded. To reload it, use:
  %reload_ext autoreload

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-5-8fb73a3658ee> in <module>()
     49 
     50 #print jacobian(f,0.)
---> 51 print jacobian(f,np.array[0.])
     52 
     53 

TypeError: 'builtin_function_or_method' object has no attribute '__getitem__'

If I changed that line into:
print jacobian(f,[0.])
the error: 
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-6-18cb7294746b> in <module>()
     50 #print jacobian(f,0.)
     51 #print jacobian(f,np.array[0.])
---> 52 print jacobian(f,[0.])
     53 
     54 # Run the simulation. Parameters not described above

/drake/lib/python2.7/site-packages/pydrake/forwarddiff.py in jacobian(function, x)
     38     The function should be vector-input and vector-output.
     39     """
---> 40     x_ad = np.empty(x.shape, dtype=np.object)
     41     for i in range(x.size):
     42         der = np.zeros(x.size)

AttributeError: 'list' object has no attribute 'shape'

So the question is, how should I use it correctly? Thanks

jwnimmer-tri, 2019-12-17:
@vinbo Per https://drake.mit.edu/getting_help.html#asking-your-question we say "If you are seeking assistance (e.g. tutorials or a brief example), please post a question on StackOverflow with the drake tag."

We'd like to use the issue tracker for bug reports; for user help, please use StackOverflow. Thank you!

closed 2019-12-17 by jwnimmer-tri

vinbo, 2019-12-22:
We need to reopen this issue I think @jwnimmer-tri . This is not about seeking a tutorial. See the discussion here: https://stackoverflow.com/questions/59438394/the-format-of-argument-of-pydrake-forwarddiff-jacobian

jwnimmer-tri, 2020-01-02:
Given the text in the top post of this issue, I cannot understand what the bug is.  If you can post a clear  statement of what the bug is (e.g., sample code, expected output, actual output), then we can reopen this.

vinbo, 2020-01-02:
> 
> 
> Given the text in the top post of this issue, I cannot understand what the bug is. If you can post a clear statement of what the bug is (e.g., sample code, expected output, actual output), then we can reopen this.

You can have a look at the stackoverflow link given above. Eric already submitted a PR about it.

PR #12511 (2020-01-02): pydrake: Use np.asarray in gradient and jacobian
Resolves #12484

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12511)