RobotLocomotion/drake #12497: second-order system example: Small tweak on tutorial code leads to a silent segfault

RussTedrake, 2019-12-18:
from stackexchange:
https://stackoverflow.com/questions/59380475/segfault-when-simulating-control-loop-with-second-order-system

I've reproduced it.   Assigning to @SeanCurtis-TRI as the current rep for the dynamics team (please delegate as necessary). 

Set priority HIGH simply because it's a pretty bad experience for new users.

SeanCurtis-TRI, 2019-12-18:
We'll look into it.

SeanCurtis-TRI, 2019-12-18:
Update:

For as yet unknown reason we're entering an infinite loop -- it looks like an undetected direct feedthrough:

```
symbolic_vector_system.cc:296            SymbolicVectorSystem<double>::CalcOutput
leaf_system.h:1772                       LeafSystem<double>::DeclareVectorOutputPort<drake::system...
std_function.h:316                       _Function_handler<void
std_function.h:706                       function<void
leaf_system.h:2622                       LeafSystem<double>::CreateVectorLeafOutputPort
std_function.h:316                       _Function_handler<void
std_function.h:706                       function<void
cache_entry.cc:52                        CacheEntry::Calc
cache_entry.h:283                        CacheEntry::UpdateValue
cache_entry.h:163                        CacheEntry::EvalAbstract
leaf_output_port.h:95                    LeafOutputPort<double>::DoEval
output_port.h:119                        OutputPort<double>::Eval<drake::AbstractValue, void>
diagram.h:1565                           Diagram<double>::EvalSubsystemOutputPort
diagram.h:1039                           Diagram<double>::EvalConnectedSubsystemInputPort
system_base.cc:183                       SystemBase::EvalAbstractInputImpl
system_base.h:135                        SystemBase::EvalAbstractInput
system.h:1888                            System<double>::DeclareInputPort
std_function.h:301                       _Function_handler<drake::AbstractValue const*
std_function.h:706                       function<drake::AbstractValue const*
input_port_base.h:82                     InputPortBase::DoEvalRequired
input_port.h:90                          InputPort<double>::Eval<drake::systems::BasicVector<doubl...
input_port.h:74                          InputPort<double>::Eval
pid_controller.cc:104                    PidController<double>::CalcControl
leaf_system.h:1772                       LeafSystem<double>::DeclareVectorOutputPort<drake::system...
std_function.h:316                       _Function_handler<void
std_function.h:706                       function<void
leaf_system.h:2622                       LeafSystem<double>::CreateVectorLeafOutputPort
std_function.h:316                       _Function_handler<void
std_function.h:706                       function<void
cache_entry.cc:52                        CacheEntry::Calc
cache_entry.h:283                        CacheEntry::UpdateValue
cache_entry.h:163                        CacheEntry::EvalAbstract
leaf_output_port.h:95                    LeafOutputPort<double>::DoEval
output_port.h:119                        OutputPort<double>::Eval<drake::AbstractValue, void>
diagram.h:1565                           Diagram<double>::EvalSubsystemOutputPort
diagram.h:1039                           Diagram<double>::EvalConnectedSubsystemInputPort
system_base.cc:183                       SystemBase::EvalAbstractInputImpl
system_base.h:135                        SystemBase::EvalAbstractInput
system.h:1888                            System<double>::DeclareInputPort
std_function.h:301                       _Function_handler<drake::AbstractValue const*
std_function.h:706                       function<drake::AbstractValue const*
input_port_base.h:82                     InputPortBase::DoEvalRequired
input_port.h:90                          InputPort<double>::Eval<drake::systems::BasicVector<doubl...
input_port.h:74                          InputPort<double>::Eval
symbolic_vector_system.cc:178            SymbolicVectorSystem<double>::PopulateFromContext<drake::...
```

We end up in a never ending cycle of each the one system attempting to evaluate the previous system until *BOOM* we run out of stack space (I crashed with roughly 64K frames).

SeanCurtis-TRI, 2019-12-18:
Looking at the code, it's obvious that the `SymbolicVectorSystem` and `PidController` both evalute their input port to compute their output value. So, there is *clearly* an algebraic loop. However, it's not being detected.

jwnimmer-tri, 2019-12-18:
`SymbolicVectorSystem::CalcOutput` is defective in that it invokes (via `PopulateFromContext`) the `get_input_port().Eval(context)` even in cases where symbolic output does not refer to input variables.

SeanCurtis-TRI, 2019-12-18:
I had come to that conclusion as well. I'm somewhat mystified why that doesn't get reported as a direct-feedthrough. The ports don't make any specific declarations about dependencies, so I would assume that they would depend on "all sources". But it nevertheless doesn't report as such.

Some vague poking suggests that the dependency tracking is superseded by analysis of the `Expression` because the feedthrough will get detected if an input variable is included in an output expression (`u0 + x0`), but goes back to being ignored if it is functionally meaningless: (`u0 * 0+ x0`).

So, I'm going to make `PopulateFromContext` smarter in that the environment gets parsed for a specific, targeted set of variables. So, each of the three calls (time derivatives, calc output, and calc discrete update will provide more specific information that should better correlate with what is reported.

PR #12500 (2019-12-19): Correct SymbolicVectorSystem "container" population
When evaluating various expressions (output, dynamics, or discrete updates), SymbolicVectorSystem builds an Environment or Substitution in order to map inputs to results.

The original implementation would conservatively attempt to populate the container with every defined variable, even if the expression being evaluated didn't need it.

This meant if there were *any* input variables, the input port would be evaluated. In small plant-controller loops, this quickly leads to spurious algebraic loops as input invokes output and vice versa foreve [... cut]
closed 2019-12-19 by soonho-tri