RobotLocomotion/drake #12498: MBP constructor should force users to pick discrete time or continuous time

RussTedrake, 2019-12-18:
After countless users have stumbled into unexpected errors because the default in MBP is continuous time, but continuous time can't handle many features like joint limits, @amcastro-tri suggested this solution.  Please make it happen?

Slack conversation: https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1555000606004800

amcastro-tri, 2020-01-02:
Per Slack discussion, it looks like we prefer `MultibodyPlant(double discrete_update_period)` to not have a default value in order to force users to always provide a sensitive value and therefore hopefully prompt them to read the accompanying documentation.
Proper deprecation would be in order.

Please provide your thoughts here if you believe there'd be better options.

amcastro-tri, 2020-01-02:
Note from @jwnimmer-tri: "Note that the parameter name time_step is part of the API, because it's bound in pydrake using that name.  I suggest not changing the name."

amcastro-tri, 2020-01-04:
Addressed by #12525.

RussTedrake, 2020-01-08:
The `AddMultibodyPlantSceneGraph` workflow, which is perhaps the more common one, still silently defaults `time_step` to zero.  
https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1multibody.html#a184cfe8f301cf07076f5707501197d88

jwnimmer-tri, 2020-01-09:
@amcastro-tri Are you working on the next PR here?  It would be nice to get it into the next tag, which is predicted to be early next week.

amcastro-tri, 2020-01-10:
per f2f with @jwnimmer-tri. We'll:
1. Provide an overloads taking `time_step` for when the plant is not provided.
2. Require a valid plant pointer for the signature taking a plant.

PR #12558 (2020-01-13): AddMultibodyPlantSceneGraph() requires time_step argument.
Second pass to resolve #12498.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12558)
closed 2020-01-13 by jwnimmer-tri