RobotLocomotion/drake #12506: URDF parser error message is not actionable

tehbelinda, 2019-12-19:
Current error messages from loading URDFs, e.g. if physically invalid, are not actionable. This makes it difficult for users to figure out what's going on in the .urdf file. For example:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  The resulting spatial inertia is not physically valid. See SpatialInertia::IsPhysicallyValid()
Aborted (core dumped)
```

I found this error when trying to load the HSR mesh from https://github.com/ToyotaResearchInstitute/hsr_meshes. Here's the example branch https://github.com/tehbelinda/drake/tree/hsr-example

amcastro-tri, 2020-01-03:
Relates to #11109.

jwnimmer-tri, 2021-11-12:
Relates to #16058.

rpoyner-tri, 2022-07-20:
I am unable to tease out any useful/reachable information from this issue that is not better expressed in #16058. Closing.

closed 2022-07-20 by rpoyner-tri