RobotLocomotion/drake #12523: `DrakeLcm` unintentionally supports regexes

ggould-tri, 2020-01-02:
`DrakeLcm::Subscribe()`, like `lcm::LCM::subscribeFunction` which it wraps, treats the channel name as a regular expression.  This allows convenient but unsupported behaviour like `Subscribe(".*", ...)` to build wrappers and decorators, which will then lead to rage when the developer attempts to use `DrakeMockLcm` for testing and discovers that it does the opposite.

We should remove the regex behaviour from `DrakeLcm` entirely (and, potentially, add a separate method for the ".*" affordance to the interface for users who might need it).

This issue was encountered in #12521, which will be updated to reference this issue in a TODO.

jwnimmer-tri, 2020-01-02:
I was going to say that we should offer a `std::regex` overload for users who want a regex... but of course `std::regex` can't echo back its pattern string because that would be too convenient.

Perhaps next best would be some kind of `std::regex::syntax_option_type` overload so the user could pass `Subscribe(std::regex::basic, ".*", &handler)` or similar.

ggould-tri, 2020-01-02:
I think the better answer would be a separate function rather than an overload -- if we have to special-case it, we should special-case it with a different handler function type that gets the actual channel name of the message.

jwnimmer-tri, 2020-01-06:
For reference, the API that lcm is using is https://developer.gnome.org/glib/stable/glib-Perl-compatible-regular-expressions.html#g-regex-match as described by https://developer.gnome.org/glib/stable/glib-Perl-compatible-regular-expressions.html#glib-Perl-compatible-regular-expressions.description -- it is a wrapper around PCRE.

We could use https://developer.gnome.org/glib/stable/glib-Perl-compatible-regular-expressions.html#g-regex-escape-string to escape the channel name.

PR #12534 (2020-01-07): lcm: Escape regex characters in DrakeLcm channel names
Drake no longer accidentally "supports" regexes for channel names.

Closes #12523.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12534)
closed 2020-01-07 by jwnimmer-tri