RobotLocomotion/drake #12528: Implement a Velocity-Implicit Euler Integrator to speed up Jacobian construction and factorization

antequ, 2020-01-03:
For second-order dynamical systems, we can implement implicit euler using Newton's method on the velocity and miscellaneous states and fixed-point iteration on the position states. This will speed up Jacobian computation and factorization. Details of this method are here: https://docs.google.com/document/d/118J6rkR9ghh2_3WY0O5EDXhfkgMOXXqTQ2usrhICv30/edit . All of this document will be transferred into doxygen or comments in the code.

This issue tracks the progress of a proposed four-PR train to merge my implementation of this integrator. 

- The first PR refactors common tests for implicit integrators.
- The second PR implements the fixed-step version with forward differencing for Jacobian computation.
- The third PR implements error control.
- The fourth PR adds autodiff and central differencing.

Tagging +@sherm1 , +@edrumwri , +@amcastro-tri for visibility

antequ, 2020-01-03:
PR #12202 contains the initial WIP. Do NOT review that PR.

sherm1, 2021-11-11:
We do have this integrator in Drake now.

closed 2021-11-11 by sherm1