RobotLocomotion/drake #12536: jupyter: Need to workaround / fix for CI flake "zmq.error.ZMQError: Address already in use"

EricCousineau-TRI, 2020-01-06:
Encountered a few times. Most recent was in #12531, in this build:
https://drake-cdash.csail.mit.edu/testDetails.php?test=37411474&build=1257354

```
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3/dist-packages/ipykernel_launcher.py", line 16, in <module>
    app.launch_new_instance()
  File "/usr/lib/python3/dist-packages/traitlets/config/application.py", line 657, in launch_instance
    app.initialize(argv)
  File "<decorator-gen-121>", line 2, in initialize
  File "/usr/lib/python3/dist-packages/traitlets/config/application.py", line 87, in catch_config_error
    return method(app, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/ipykernel/kernelapp.py", line 456, in initialize
    self.init_sockets()
  File "/usr/lib/python3/dist-packages/ipykernel/kernelapp.py", line 243, in init_sockets
    self.stdin_port = self._bind_socket(self.stdin_socket, self.stdin_port)
  File "/usr/lib/python3/dist-packages/ipykernel/kernelapp.py", line 180, in _bind_socket
    s.bind("tcp://%s:%i" % (self.ip, port))
  File "zmq/backend/cython/socket.pyx", line 495, in zmq.backend.cython.socket.Socket.bind (zmq/backend/cython/socket.c:5662)
  File "zmq/backend/cython/checkrc.pxd", line 25, in zmq.backend.cython.checkrc._check_rc (zmq/backend/cython/socket.c:8400)
zmq.error.ZMQError: Address already in use
Running notebook as a test (non-interactive):
  mathematical_program_multibody_plant.ipynb
Traceback (most recent call last):
  File ".../bin/tutorials/py/mathematical_program_multibody_plant_test.runfiles/drake/tutorials/mathematical_program_multibody_plant_jupyter_py_main.py", line 14, in <module>
    main()
  File ".../bin/tutorials/py/mathematical_program_multibody_plant_test.runfiles/drake/tutorials/mathematical_program_multibody_plant_jupyter_py_main.py", line 10, in main
    _jupyter_bazel_notebook_main(cur_dir, notebook, sys.argv[1:])
  File ".../bin/tutorials/py/mathematical_program_multibody_plant_test.runfiles/drake/tools/jupyter/jupyter_bazel.py", line 60, in _jupyter_bazel_notebook_main
    ep.preprocess(nb, resources={'metadata': {'path': notebook_dir}})
  File "/usr/lib/python3/dist-packages/nbconvert/preprocessors/execute.py", line 257, in preprocess
    cwd=path)
  File "/usr/lib/python3/dist-packages/nbconvert/preprocessors/execute.py", line 241, in start_new_kernel
    kc.wait_for_ready(timeout=startup_timeout)
  File "/usr/lib/python3/dist-packages/jupyter_client/blocking/client.py", line 120, in wait_for_ready
    raise RuntimeError('Kernel died before replying to kernel_info')
RuntimeError: Kernel died before replying to kernel_info
```

From looking at source near failure:
https://github.com/ipython/ipykernel/blob/4.8.2/ipykernel/kernelapp.py#L180

Seems like we can possibly configure `IPKernelApp.stdin_port` and set it to 0, so that it can be selected at random, at least during testing?

EricCousineau-TRI, 2020-01-06:
Hm... Seems like the default should be 0 (random port)?
https://ipython.readthedocs.io/en/stable/config/options/kernel.html#configtrait-IPKernelApp.stdin_port

EricCousineau-TRI, 2020-05-07:
One workaround is to hack in config override when doing `bazel test`:
https://github.com/RobotLocomotion/drake/blob/6761809c64237d54ac8a8bb37d4daccdbe21fb55/tools/jupyter/jupyter_bazel.py#L46-L49

BetsyMcPhail, 2020-05-20:
There is an [nbconvert issue](https://github.com/jupyter/nbconvert/issues/1066) that *may* have fixed this problem. It suggests to upgrade to:

**jupyter_client  5.3.4
ipykernel 5.1.2**

Currently, on Bionic:
**jupyter_client  5.2.2
ipykernel 4.8.2**

On Focal:
**jupyter_client  6.1.2
ipykernel 5.2.0**

On Mac, both Catalina and Mojave:
**jupyter_client  6.1.3
ipykernel 5.2.1**

EricCousineau-TRI, 2020-05-21:
Thanks! So for this route, we the options then are to:

1. Wait until we drop support for Bionic, since Focal and Homebrew
2. Vendor in `jupyter notebook` on Ubuntu (eww)
3. Try some other alt.

I think Option 1 makes the most sense. I will briefly tinker with 3.

FTR, I can generally reproduce these failures on my laptop (8 cores) by simulating "Jupyter overload" (in some runs, 1 / 200 fail, in others, 3 / 200):
```sh
bazel test \
  --flaky_test_attempts=1 --test_sharding_strategy=disabled --test_summary=short \
  --jobs 10 --runs_per_test=200 \
  //tools/jupyter:py/example_test
```

EricCousineau-TRI, 2020-05-21:
Tried tinkering with jupyter config, but still can't navigate that complex labyrinth to have it use the config and print out more debug info (to see if the config is correct, and it's truly the race condition):

https://github.com/RobotLocomotion/drake/compare/master...EricCousineau-TRI:issue-12536-repro (3189bc164a)

BetsyMcPhail, 2020-05-22:
I agree that 1. seems reasonable.

EricCousineau-TRI, 2020-05-22:
Filed #13391, which is a prereq for this.

ToffeeAlbina-TRI, 2020-05-28:
@EricCousineau-TRI should we just close this as won't fix?

EricCousineau-TRI, 2020-05-28:
Sure.

closed 2020-05-28 by EricCousineau-TRI

jwnimmer-tri, 2021-11-04:
If https://github.com/bazelbuild/bazel/issues/13588 ever appears, we can use it so that ZMQ errors are retried, but real errors still in the code under test produce an immediate failure.